RetroPie/RetroPie-Setup #1088: lr-mednafen-pce is requesting my github login and password when compiling from source

ProxyCell, 2015-11-17:
Log started at: Tue Nov 17 16:27:27 EST 2015
/home/pi/RetroPie-Setup/tmp/build/lr-mednafen-pce /home/pi/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-mednafen-pce' : PCEngine emu - Mednafen PCE core port for libretro
= = = = = = = = = = = = = = = = = = = = =

git clone --depth 1 "https://github.com/petrockblog/mednafen-pce-libretro.git" "/home/pi/RetroPie-Setup/tmp/build/lr-mednafen-pce"
Cloning into '/home/pi/RetroPie-Setup/tmp/build/lr-mednafen-pce'...
Username for 'https://github.com':

gizmo98, 2015-11-18:
@petrockblog has no mednafen-pce-libretro repo (anymore?).

joolswills, 2015-11-18:
It's possible I removed it as we no longer needed our fork due to upstream changes. Can't remember but we can switch to the upstream repo.

joolswills, 2015-11-18:
oh wait. sorry - nothing to do with me - misread and though it was on the RetroPie repository.

ProxyCell, 2015-11-18:
I'm not sure if this matters but it is a RPI2 running Wheezy (I gave up on Jessie for now...)
Is this just being given the wrong git repo URL?

joolswills, 2015-11-18:
the repository is removed - I guess @petrockblog removed it - his repository may have had changes from upstream I guess but we can find the upstream repo and use that (if it works).

ProxyCell, 2015-11-18:
the "fast" variant of this is still there

HerbFargus, 2015-11-18:
If this was the upstream, https://github.com/libretro/mednafen-pce-libretro
It no longer exists. Probably superceded by pce-fast.

ProxyCell, 2015-11-18:
thanks @HerbFargus :+1:  but google and archive dot org do not have a cache of it :(

joolswills, 2015-11-18:
then perhaps we can just remove the old module. I will look into it further when I get a moment. there may be some other alternative libretro emulators too.

joolswills, 2015-11-18:
@ProxyCell for now just install from binary and don't worry about it.

ProxyCell, 2015-11-18:
:+1: 

Thanks @joolswills

petrockblog, 2015-11-18:
I am sorry for that! Without double checking it I assumed that that fork is not needed anymore. I will try to find an alternative ...

ProxyCell, 2015-11-18:
thanks @petrockblog 
its nothing of critical concern but it is different from the fast variant as it focusses on accuracy over speed - i figure its worth letting people switch between the two if there is a need based upon the game/software

joolswills, 2015-11-18:
If it is unsupported upstream, it may have been removed for a reason (ie not being needed). I am happy to remove the module unless someone has evidence over better accuracy etc vs the other libretro core.

joolswills, 2015-11-18:
I found another fork of it, but it dates back to 2012. If it is not being maintained by libretro devs anyway, I don't see the point in keeping it. I'm going to remove the module.

ProxyCell, 2015-11-18:
@joolswills i agree, i'm not familiar with anything that REQUIRES this but i had been using it for quite a while over the FAST

petrockblog, 2015-11-18:
I also found a fork from 2012 - probably the same. I have forked that now so that the module would work. But we could also remove the module. It is not that we are really loosing functionality in RetroPie without that ...

joolswills, 2015-11-18:
>me: so mednafen-pce-libretro became beetle-pce-fast-libretro there were never two ports ?
> <retroarch dev> correct

closed 2015-11-18 by joolswills

ProxyCell, 2015-11-18:
this is crazy, we've had two for ages and they were entirely the same?

PR #3724 (2023-06-14): amiberry: update to 5.6.1
Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation. Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #108 [... cut]
commit 042ae93c (2023-06-14): amiberry: update to 5.6.1

Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation.
Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #10 [... cut]