RetroPie/RetroPie-Setup #1089: Add [Theme] - retroplay_clean_canela

InsecureSpike, 2015-11-18:
hi guy's would you like to review my new theme for addition to the theme installer?

https://github.com/InsecureSpike/retroplay_clean_canela

small file size (under 6MB)
clean & clear looking
16:9 & 4:3 aspect ratio compatible!
interchangeable logo banners
34 alternate & extra logo banners availabe @ https://insecurespike.wordpress.com/the-downloads/

screenshot viewable here https://insecurespike.wordpress.com/clean_canela-screenshots/

thank you

InsecureSpike, 2015-11-18:
just replaced the "RetroPlay" text wallpaper with a no text wallpaper

joolswills, 2015-11-18:
For themes to be added to our theme installer your repository needs to be named like `es-theme-simple` eg `es-theme-retroplay-clean-canela`

InsecureSpike, 2015-11-18:
oh sorry, are they ok now?

joolswills, 2015-11-18:
yep. Ill test / add them when I next do some development. thanks.

InsecureSpike, 2015-11-18:
awesome thank you

InsecureSpike, 2015-11-18:
great! both are now ready whenever someone has some time
thanks

HerbFargus, 2015-11-18:
https://github.com/RetroPie/RetroPie-Setup/pull/1091/files

joolswills, 2015-11-18:
@HerbFargus have you tested them ?

HerbFargus, 2015-11-19:
I haven't tested the latest changes, I'll test within the hour though and report back

HerbFargus, 2015-11-19:
They work fine, though I think the clean version is supposed to have a main system wallpaper? Right now it just shows up as a white background. The gamelist wallpaper shows up fine. @InsecureSpike do you want to check and make sure this theme is working as you intended? Regardless, minor details can be fixed upstream and it can still be merged I think.

HerbFargus, 2015-11-19:
Yeah you're missing the clean_mainmenu.png on this theme, otherwise it seems fine.

InsecureSpike, 2015-11-19:
sorry, the wallpaper is there now, thank you

HerbFargus, 2015-11-19:
They have been added, you can update your setup script and they should be there- so you should be able to close your two tickets now.

closed 2015-11-19 by InsecureSpike