RetroPie/RetroPie-Setup #1097: Can't compile experimental Emulator lr-desmume

Melcor85, 2015-11-25:
Hi people,

I'm trying to compile the experimental lr-desmume via the menu that shows up by entering "sudo ./retropie_setup.sh" after updating the retropie setup script.

I just get errors ("invalid use of 'this' in non-member function") and at the end the error "/home/pi/RetroPie-Setup
Could not successfully build NDS emu - DESMUME (/home/pi/RetroPie-Setup/tmp/build/lr-desmume/desmume/desmume_libretro.so not found)".

I've already read those repositories
https://github.com/libretro/desmume/commit/8d865b402d8d0752d22f82c7e5a94e8ec69f7709
https://github.com/libretro/desmume/issues/61
https://github.com/RetroPie/RetroPie-Setup/issues/1030

but i'm a noob to linux and have just rudimentary knowledge about programming etc. and so I don't get the solution.

Here is the log file:
http://pastebin.com/2NhVZDWq

I'm thankfull for every help.

Greetings Melcor85

joolswills, 2015-11-25:
We had a workaround for https://github.com/libretro/desmume/issues/61 in the module. it isn't needed now and the patch fails, so it won't builld. I have removed it and it should build now (update retropie-setup)

Melcor85, 2015-11-25:
Thank you very much. It compiled successfull!

closed 2015-11-25 by Melcor85