RetroPie/RetroPie-Setup #1100: Add reicast input script

gizmo98, 2015-11-26:
Reicast is not really usable at the moment. There is no in game setup menu and joypad config looks a little bit strange. I created a input script, so joypad setup shouldn't be complicated anymore:
https://github.com/gizmo98/RetroPie-Setup/blob/8c5f3bbf7cb09f91b85c0de41d115540dd42ae17/scriptmodules/supplementary/emulationstation/configscripts/reicast.sh

Reicast seems to have no hat support. Just map any left axis device to analog stick if possible. Quit emulator with "select".

HerbFargus, 2015-11-26:
I'll give it a go when I've got time today

gizmo98, 2015-11-26:
Thanks!

I don't know the state of reicast. I had no problem starting games. Is there a reason we keep reicast experimental?

HerbFargus, 2015-11-26:
This is the latest on it:

https://github.com/RetroPie/RetroPie-Setup/issues/897

HerbFargus, 2015-11-26:
Alright a few thoughts: All in all Quite Brilliant!

I have been testing on my android with reicast and with the raspberry pi to compare the original controls and I'm wondering if it would be prudent for consistencie's sake to match the controls based on retroarch's default snes method. for example with my logitech f310 I map it according to the following image:

https://cloud.githubusercontent.com/assets/10035308/7110173/0f2ea784-e16a-11e4-9c6f-5fe7c594b05a.png

So basically that would entail switching (A and B), and ( X and Y )

For the F310 the dpad of course doesnt work as it has hats, 

For my Snes controller the dpad works fine. 

the top shoulder buttons dont work on either of my controllers, not sure if it doesnt work on my f310 because its an axis (the same buggy issue with a lot of other things) or if its just something in the configs. 

having select be quit is a great idea- definitely beats ctrl+c

HerbFargus, 2015-11-26:
I found this mapping to be more intuitive for me:

```

#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

function onstart_reicast_joystick() {
    local device_type=$1
    local device_name=$2

    # write temp file header
    echo "; ${device_name}_START " > /tmp/reicasttempconfig.cfg
    echo "[${device_name}]" >> /tmp/reicasttempconfig.cfg
    iniConfig " = " "" "/tmp/reicasttempconfig.cfg"
}

function map_reicast_joystick() {
    local device_type="$1"
    local device_name="$2"
    local input_name="$3"
    local input_type="$4"
    local input_id="$5"
    local input_value="$6"

    local keys
    local dir
    case "$input_name" in
        up)
            keys=("DPad_Up")
            ;;
        down)
            keys=("DPad_Down")
            ;;
        left)
            keys=("DPad_Left")
            ;;
        right)
            keys=("DPad_Right")
            ;;
        b)
            keys=("Btn_A")
            ;;
        y)
            keys=("Btn_X")
            ;;
        a)
            keys=("Btn_B")
            ;;
        x)
            keys=("Btn_Y")
            ;;
        leftbottom)
            keys=("Axis_LT")
            ;;
        rightbottom)
            keys=("Axis_RT")
            ;;
        lefttop)
            keys=("DPad2_Left")
            ;;
        righttop)
            keys=("DPad2_Right")
            ;;
        start)
            keys=("Btn_Start")
            ;;
        select)
            keys=("Quit")
            ;;
        leftanalogleft)
            keys=("Axis_X")
            ;;
        leftanalogright)
            keys=("Axis_X")
            ;;
        leftanalogup)
            keys=("Axis_Y")
            ;;
        leftanalogdown)
            keys=("Axis_Y")
            ;;
        *)
            return
            ;;
    esac

    local key
    local value
    #iniConfig " = " "" "/tmp/mp64keys.cfg"
    for key in "${keys[@]}"; do
        # read key value. Axis takes two key/axis values.
        case "$input_type" in
            axis) 
                # key "X/Y Axis" needs different button naming
                if [[ "$key" == "Axis_X" || "$key" == "Axis_Y" ]] ; then
                    iniSet "axis.$input_id" "${key}" 
                elif [[ "$key" == "DPad_Up" || "$key" == "DPad_Down" ]]; then
                    iniSet "axis.$input_id" "Axis_Y" 
                elif [[ "$key" == "DPad_Right" || "$key" == "DPad_Left" ]]; then
                    iniSet "axis.$input_id" "Axis_X" 
                fi
                ;;
            hat)
                ;;
            *)
                iniSet "button.$input_id" "$key" 
                ;;
        esac
    done
}

function onend_reicast_joystick() {
    local device_type=$1
    local device_name=$2

    echo "; ${device_name}_END " >> /tmp/reicasttempconfig.cfg
    echo "" >> /tmp/reicasttempconfig.cfg

    # abort if old device config cannot be deleted.
    # keep original mupen64plus-input-sdl configs.
    local file="/home/pi/.reicast/emu.cfg"
    if [[ ! -d "/home/pi/.reicast" ]]; then
        mkdir /home/pi/.reicast
    fi
    if [[ -f "$file" ]]; then
        # backup current config file
        cp "$file" "${file}.bak"
        # if reicast did not run frames are there
        if grep -q "${device_name}_END" "$file" ; then
            sed -i /"${device_name}_START"/,/"${device_name}_END"/d "$file"
        # reicast removes frames after first run but adds axis.31
        else
            sed -i /"${device_name}"/,/"axis.31="/d "$file"
        fi
        if grep -q "$device_name" "$file" ; then
            rm /tmp/reicasttempconfig.cfg
            return
        fi
    fi

    # read temp device configuration and append to InputAutoCfg.ini
    local deviceconfig=`cat /tmp/reicasttempconfig.cfg`
    echo "$deviceconfig" >> "$file"
    rm /tmp/reicasttempconfig.cfg
}
```

HerbFargus, 2015-11-26:
you may also want to clean up the copy and paste comments for mupen64 ;) 
https://github.com/gizmo98/RetroPie-Setup/blob/8c5f3bbf7cb09f91b85c0de41d115540dd42ae17/scriptmodules/supplementary/emulationstation/configscripts/reicast.sh#L125

gizmo98, 2015-11-27:
Ok. i changed A/B/X/Y mapping. It could be L/R axis trigger mapping works now. @HerbFargus could you take a look if your shoulder buttons of F310 work?

HerbFargus, 2015-11-27:
did you remap them to left bottom and right bottom?  I just tested with my F310 and they seem to be working

gizmo98, 2015-11-27:
I shoulder button is a axis i add a real LT/RT axis now. The first draft added X/Y axis only ;-)

HerbFargus, 2015-11-27:
cool - well as far as it stands now I don't see any other issues, all looks good to me :+1:

gizmo98, 2015-11-27:
Ok. I have done a PR.

closed 2015-11-27 by gizmo98

robertybob, 2015-11-27:
@gizmo98 How does this work? If I install Recast now (for the first time) will it pick up my existing Xbox 360 controller config (from Emulationstation)?

gizmo98, 2015-11-28:
It ist necessary to run Input Setup in emulationstation for your Xbox Controller again.

PR #3724 (2023-06-14): amiberry: update to 5.6.1
Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation. Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #108 [... cut]
commit 042ae93c (2023-06-14): amiberry: update to 5.6.1

Among others, fixes some input issues, one of them reported in the forums, and a Vero4k black screen during emulation.
Full changelog at https://github.com/BlitterStudio/amiberry/releases/tag/v5.6.1

Bugfixes:

- Hardfile properties were always set to default (fixes #1077)
- cycle exact was incorrect enabled sometimes (fixes #1075)
- Fix foreign disk image support when drive is standard Amiga 3.5" DD
- hsync events should be before misc events
- fixed uaegfx board parameters
- uaegfx board name is UAE, not uaegfx.card
- Fix helptext formatting in CPU panel (fixes #10 [... cut]