RetroPie/RetroPie-Setup #1101: Mupen64Plus Controller config - A / B buttons recognized as dpad buttons?

TheReal1604, 2015-11-26:
Hey Guys,

got some problem with a fresh retropie installation...

For psx and snes the controller config works just fine, but for n64 i have the strange issue that the a / b / x / y button is recognized as dpad left / right / up / down.. in Super Smash bros. 

Did the reconfiguration via script.. have tried to set the controls via Inputautocfg.ini .. just not working. (Checked the button numbers before with jstest)

Got a Dragonrise controller.

Anyone a idea how i can fix this?

Greetz
TheReal

HerbFargus, 2015-11-26:
make a backup, delete Inputautocfg.ini, update your setup script, do a full binary install, reconfigure your controllers, and then see if it works.

TheReal1604, 2015-11-26:
Thx for your very fast response. What is the preferred way to rebuild the controller config? Retro pie-Setup Script?

HerbFargus, 2015-11-26:
its automatically done through the emulationstation controller config now after a full binary update. just added yesterday

TheReal1604, 2015-11-27:
Hey HerbFargus,

with your steps, it worked now! Thanks for that!

Additionally i configured the gles2n64 as default video plugin.. after your steps i now have no sound in n64... any idea? :-P

gizmo98, 2015-11-27:
@TheReal1604 
Audio plugin does not auto detect audio output path. To use 3,5mm audio jack you have to modify mupen64plus.cfg.

gizmo98, 2015-11-27:
I think this can be closed now.

closed 2015-11-27 by gizmo98