RetroPie/RetroPie-Setup #1104: [mupen64plus] No sound and strange keymap when using rice

reynum2, 2015-11-28:
Hello,

I use 
```bash
pi@retropie ~ $ uname -a
Linux retropie 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux
```
On a Raspberry Pi 2

I tried to play a n64 game, the game launch correctly the sound works and the buttons are at the good place. But there was lots of lags and graphical bugs, so i decided to use Rice plugin. I read the advise here : http://www.mupen64plus.org/wiki/index.php?title=GameCompatibility
[image]
After than the video was perfect but no sound at all and the buttons of gamepad were shuffle (and some disappear)

I tried to modify the file /opt/retropie/configs/n64/mupen64plus.cfg

from
```
# Digital button configuration mappings
DPad R = "axis(5+)"
DPad L = "axis(5-)"
DPad D = "axis(6+)"
DPad U = "axis(6-)"
Start = "button(9)"
Z Trig = "button(7)"
B Button = "button(8)"
A Button = "button(6)"
C Button R = "button(1)"
C Button L = "button(3)"
C Button D = "button(2)"
C Button U = "button(0)"
R Trig = "button(5)"
L Trig = "button(4)"
Mempak switch = "key(109)"
Rumblepak switch = "key(114)"
# Analog axis configuration mappings
X Axis = "axis(0-,0+)"
Y Axis = "axis(1-,1+)"
```
to
```
# Digital button configuration mappings
DPad R = "axis(5+)"
DPad L = "axis(5-)"
DPad D = "axis(6+)"
DPad U = "axis(6-)"
Start = "button(9)"
Z Trig = "button(7)"
B Button = "button(3)"
A Button = "button(2)"
C Button R = "axis(2+)"
C Button L = "axis(2-)"
C Button D = "axis(3+)"
C Button U = "axis(3-)"
R Trig = "button(5)"
L Trig = "button(4)"
Mempak switch = "key(109)"
Rumblepak switch = "key(114)"
# Analog axis configuration mappings
X Axis = "axis(0-,0+)"
Y Axis = "axis(1-,1+)"
```
but each time I launch the game theses values reset at default.

I also tried to change the values in the file /opt/retropie/configs/n64/InputAutoCfg.ini
```ini
[DragonRise Inc. Generic USB Joystick]
plugged = True
plugin = 2
mouse = False
AnalogDeadzone = 4096,4096
AnalogPeak = 32768,32768
DPad R = key(275) axis(0+)
DPad L = key(276) axis(0-)
DPad D = key(274) axis(1+)
DPad U = key(273) axis(1-)
Start = button(9)
Z Trig = button(7)
B Button = button(3)
A Button = button(2)
C Button R = axis(2+)
C Button L = axis(2-)
C Button D = axis(3+)
C Button U = axis(3-)
R Trig = button(5)
L Trig = button(4)
Mempak switch =
Rumblepak switch =
X Axis = axis(0-,0+)
Y Axis = axis(1-,1+) 
```
But it changes nothing

For the sound I tried to change the audio plugin 
```
[UI-Console]

# Mupen64Plus UI-Console config parameter set version number.  Please don't change this version number.
Version = 1
# Directory in which to search for plugins
PluginDir = "./"
# Filename of video plugin
VideoPlugin = "mupen64plus-video-rice.so"
# Filename of audio plugin
AudioPlugin = "mupen64plus-audio-omx.so"
# Filename of input plugin
InputPlugin = "mupen64plus-input-sdl.so"
# Filename of RSP plugin
RspPlugin = "mupen64plus-rsp-hle.so"
```
I change the line to : AudioPlugin = "mupen64plus-audio-sdl.so"
The audio plugin (located in /opt/retropie/emulators/mupen64plus/lib/mupen64plus) works I have sound but no image :-s.
Note : This change is not erase when I load the game.

Any idea to have Ocarina of time with good graphics, sound and working gamepad ? ;-)

gizmo98, 2015-11-28:
Use GLideN64. Rice does not work with Audio-SDL. 

If you have still controller issues read this:
https://github.com/RetroPie/RetroPie-Setup/issues/1101

reynum2, 2015-11-28:
Thanks with two lines you unlock me :-)
The GLide plugin don't have the same graphics quality than Rice but works well with sound !
The link about the gamepad issue is good too. indeed in the InputAutoCfg.ini it wasn't the _DragonRise Inc. Generic USB Joystick_ (my controller) who was loaded but the _Generic   USB  Joystick_ keymap so I have modify this one and all works well :-D

closed 2015-11-28 by reynum2

alwe2710, 2016-04-10:
You were nearly right.
go to `/opt/retropie/configs/n64/` and edit `mupen64plus.cfg`

And the second paragraph should look like this:
`[Audio-OMX]

# Mupen64Plus OMX Audio Plugin config parameter version number
Version = 1
# Frequency which is used if rom doesn't want to change it
DEFAULT_FREQUENCY = 33600
# Swaps left and right channels
SWAP_CHANNELS = False
# Audio output to go to (0) Analogue jack, (1) HDMI
OUTPUT_PORT = 0
# Audio Output Frequncy mode (when NATIVE_MODE=false): 0 = Rom Frequency, 1 ROM Frequency if supported (HDMI only), 2 = Standard frequency < Rom Frequency, 3 = Standard frequency > Rom Frequency, [N] Force output frequency
DEFAULT_MODE = 0
`

gizmo98, 2016-04-10:
Do a source based installation of mupen64plus. This problem does only happen with older rice versions.

joolswills, 2016-04-10:
Also please use the forum first in the future for configuration help - this issue tracker is for bugs (and if not sure, always best to post to the forum to get feedback)

joolswills, 2016-04-10:
wrong thread :)