RetroPie/RetroPie-Setup #1111: Megadrive platform no longer exists (for scraping)

HerbFargus, 2015-12-03:
Is there a way to change the platform for es_systems.cfg to be genesis when the configs are created since there is no longer a scraping platform for megadrive?

see http://thegamesdb.net/platforms/

http://blog.petrockblock.com/forums/topic/cant-scrape-megadrive-no-games-found/

joolswills, 2015-12-03:
I would rather they fix it so you can look up with the megadrive name. Have you spoken to thegamesdb.net ? :)

HerbFargus, 2015-12-03:
well they did have it, and they removed it. I'll log a post there and see what they say.

joolswills, 2015-12-03:
Are you referring to the built in ES scraper btw ?

joolswills, 2015-12-03:
if we did fix it our end, the best bet would be to change the scraper in ES for thegamesdb to use genesis for megadrive rather than changing around the way we generate the ES config and having to redo all the themes etc.

HerbFargus, 2015-12-03:
Yeah, simplest is best.  (I think it applies to both Sselphs and the built in- not entirely sure though)

here's the post: http://forums.thegamesdb.net/topic/43/add-megadrive-as-its-own-platform

joolswills, 2015-12-03:
i think you asked the wrong question. They were two platforms in their system and they merged megadrive with genesis. What we need is so when we query with megadrive we get data back (even it if it genesis). That would be useful - so like a platform alias etc.

HerbFargus, 2015-12-03:
http://forums.thegamesdb.net/topic/43/add-platform-alias-for-merged-platforms

joolswills, 2015-12-03:
Looks good :) Let's see what they say.

HerbFargus, 2015-12-04:
I guess they have suggested regional artwork flags http://forums.thegamesdb.net/topic/19/thegamesdb-site-ideas-and-improvements so its just a matter of getting those implemented (if they ever are implemented) and then adapting our current scrapers to match the API. Though we would still probably have to change some platform flags when they merge the systems.

HerbFargus, 2016-02-22:
Not much we can do about this as it pends on thegamesdb so closing for now.

closed 2016-02-22 by HerbFargus