RetroPie/RetroPie-Setup #1113: Option to disable/enable automapping (reicast, mupen64plus...)

gizmo98, 2015-12-04:
I would like to add startup controller mapping scripts for fba, ppsspp, mame and some other emulators which use static input setups. The user should not be patronized so there must be an option to opt out this functionality. I don't know if a global retroarch.cfg is appropriate. Here is an example code: https://github.com/gizmo98/RetroPie-Setup/commit/9157a0281d093c881d7c9a71127b53fd2534d7de

HerbFargus, 2015-12-04:
I take it you are discussing both the autoconfig scripts as well as the hotkey scripts? I think it's reasonable to give users the option for those that want their own configurations.

I know @joolswills had discussed wanting to integrate hotkeys into the autoconfiguration script in emulationstation 

And as far as PiFBA there are some input bugs that would also need to be addressed https://github.com/RetroPie/RetroPie-Setup/issues/908 

I also know that there may be some issues with PPSSPP configs (as far as the menu is concerned) when I configured my controller I was unable to navigate the menu properly, so that may also need to be addressed as well.

gizmo98, 2015-12-04:
If we add an automatism of any kind (input, hotkeys, video, audio) some users will be certainly confused or disappointed because they like there self made own configurations. This is no acute issue. I would see  opt out option for automatisms as an enhancement.

You cannot use autoconfiguration script "only" if the emulator only has a static input setup like reicast upstream. Reicast upstream needs a startup helper script to link a controller related mapping files to your connected controllers.

joolswills, 2015-12-04:
I'm happy to have a feature to disable the autoconfiguration, although let's not call it retropie.cfg, but something specific for this - we don't have a central "retropie" configuration (eg dispmanx settings are in own config, runcommand has a few configs, so it would be more consistent to have it named autoconfig.cfg or something perhaps)

joolswills, 2015-12-04:
it would make sense to have a function to handle it rather than duplicating similar lines in each module also.

joolswills, 2015-12-04:
I would prefer 0 / 1 instead of true / false also.

gizmo98, 2015-12-05:
Ok. I will take a look something like this can be implemented.
-store settings in autoconfig.cfg (for every system or in $configs/all?)
-add two functions. getAuto "if something do something" and addAuto "add something if nothing set". 
-use 0/1 to enable/disable option

joolswills, 2015-12-05:
I think in all - so it's easily processed by a module gui etc.

sounds good.

gizmo98, 2015-12-06:
Ok i have two functions now. Where can i store them? inifuncs.sh:
```
#!/usr/bin/env bash
rootdir="/opt/retropie"
configdir="$rootdir/configs"

source "$rootdir/lib/inifuncs.sh"

function addAutoConf() {
    local value="$1"
    local default="$2"

    if [[ -z "$default" ]]; then
       eco "no default"
       default="1"
    fi

    iniConfig " = " "" "$configdir/all/autoconf.cfg"
    iniGet "$value"
    ini_value="${ini_value// /}"
    if [[ -z "$ini_value" ]]; then
        iniSet "$value" "$default"
    fi
}

function getAutoConf(){
    local value="$1"

    iniConfig " = " "" "$configdir/all/autoconf.cfg"
    iniGet "$value"

    if [[ "$ini_value" == "1" || "$ini_value" == "0" ]]; then
        return "$ini_value"
    fi
    return 0
}

addAutoConf test 1
getAutoConf test || echo "do something"
```

joolswills, 2015-12-06:
Yeh you can include them in there if they are to be used in inputconfiguration and modules etc.

closed 2015-12-07 by gizmo98