RetroPie/RetroPie-Setup #1118: Documentation, FAQs, Guides, etc [Feature Request]

ProxyCell, 2015-12-05:
Howdy,

Would it be a good idea to add some kind of documentation/guides/FAQs/etc into RetroPie itself? Possibly accessible from Emulation Station?

I know this is kind of crazy-sounding but if we can agree on a way to load/display them on screen with just a gamepad I don't think its so bad.

robertybob, 2015-12-06:
One idea I had was to have add a help section, perhaps next to RetroPie within Emulationstation. Within that are shortcuts to PDFs for different things (adding roms, installing new themes etc) - there are a few pdf readers for Raspbian and I figure this would be quicker and easier than linking to a website or GUI screen with help info.

joolswills, 2015-12-06:
In terms of the additional work to add and maintain this I prefer it as it is currently with the documentation online.

HerbFargus, 2015-12-06:
I would prefer to only maintain one set of documentation. Perhaps if RetroPie were more static and wasn't in constant development, it would be more viable.

joolswills, 2015-12-06:
I did a bit of reading earlier, and I may consider this as a retropie module if we have it so we checkout the documentation repository,  and then provide a simple console render of the markdown documentation - that way we don't need to maintain two sets of documentation. However, I have other things I want to do first.

joolswills, 2015-12-06:
eg. using the "markdown" command, and viewing in lynx on terminal.

HerbFargus, 2015-12-06:
Sounds doable, images might be a bit of a trick.

joolswills, 2015-12-06:
yeh it wouldn't include images with this method, but it would be simple and easy to update from the rpi.

ProxyCell, 2015-12-07:
oh wow, you guys have been really running with this idea I see
i just felt it would be nice to have something offline and accessible for users of the device itself

i was even thinking of creating some myself so you guys didn't have to maintain anything new. i was really thinking along the lines of just a few articles/FAQs for features like the SAMBA share, USB roms, etc

HerbFargus, 2016-06-19:
this is more or less integrated.

closed 2016-06-19 by HerbFargus