RetroPie/RetroPie-Setup #1121: Wrong button mapping when raspi-config launched from retropie menu

ali1234, 2015-12-07:
To reproduce:

1. Start up retropie, see emulation station menu.
2. Go to retropie menu, then select raspi-config option.

When in raspi-config, I must press "B" to activate the select/finish buttons. In other such text-mode config screens it is done with the "A" button.

ScOULaris, 2015-12-10:
I have noticed this as well.

tynor88, 2015-12-14:
Yes me too, minor bug :-)

joolswills, 2015-12-14:
Not a bug - we use a joy2key script to convert joypad to keypresses. As raspi-config uses a different console gui from RetroPie-Setup it has different controls. This won't be changed - and for this sort of configuration a keyboard is still easier.

joolswills, 2015-12-14:
raspi-config is not developed by us either.

closed 2015-12-14 by joolswills