RetroPie/RetroPie-Setup #1122: Reboot option halts/hangs the system when raspi-config is launched from menu

ali1234, 2015-12-07:
To reproduce:

1. In EmulationStation go to retro pie menu and launch raspi-config
2. In raspi-config select the "reboot system" option (last one in the list.)

Expected result: system should reboot.
Actual result: System halts/hangs and I must power cycle it.

System: Raspberry Pi 2 with official 7" display, retropie 3.2.1.

joolswills, 2015-12-07:
As we don't maintain raspi-config there is nothing I can do about that really, just use the reboot from within emulationstation.

ali1234, 2015-12-07:
It's been my experience that the reboot option of raspi-config works correctly when run from within an official raspbian image...

HerbFargus, 2015-12-07:
Though the reboot function seems to work fine after running expand filesystem etc.

joolswills, 2015-12-07:
I don't see a "reboot system" option in raspi-config (apart from after you expand filesystem etc which works)

joolswills, 2015-12-07:
retropie-setup has "Perform Reboot" as the last option, which works fine here also.

ali1234, 2015-12-07:
Yes, it was after I resized the filesystem. Didn't realize the option wasn't there before. Anyway, it didn't work for me. The system halted instead of rebooting.

joolswills, 2015-12-07:
it reboots here - maybe you got unlucky or something. No-one has mentioned this before and this something everyone runs after install. You could try and reproduce it, perhaps something related to your hardware setup etc.

HerbFargus, 2015-12-07:
Possible bad SD card, power supply, too aggressive overclock etc.

joolswills, 2015-12-08:
ok to close ?

closed 2015-12-10 by joolswills