RetroPie/RetroPie-Setup #1125: PS3 Controllers Default to Players 3/4 When Xbox Driver Is Also Installed

Stunan13, 2015-12-11:
Issue:
PS3 controllers are always assigned to players 3/4 when connecting to the retropie. 

The retropie should be able to dynamically assign wireless inputs to players in the order that they connect via bluetooth including a mix of input types e.g. player 1 - PS3 Controller, player 2 - Xbox controller, player 3 - 8bitdo controller etc...

Reproduction Steps:
1: Install the PS3 Controller driver
2: Pair 2 PS3 controllers via bluetooth
3: Install the xbox controller driver

Workaround:
Uninstall the xbox controller driver via sudo apt-get remove xboxdrv

joolswills, 2015-12-11:
you may have some luck editing the xboxdrv config in /etc/rc.local - You may be able to configure it so it works as you want for your setup but it might need hardcoding for how you want your controllers connected. These are 3rd party applications, and although we have some improvements in the ps3 driver, the xboxdrv is not developed by us. We also ship with an older version currently. See documentation here http://pingus.seul.org/~grumbel/xboxdrv/xboxdrv.html

joolswills, 2016-01-09:
xboxdrv has been updated and has been improved somewhat, but you still would need to manually configure the xboxdrv if you want it only used for id 0/2 etc

joolswills, 2016-03-26:
I dont think it's realistic to ever make these two drivers work together dynamically in a way that would work for all set ups. Closing.

closed 2016-03-26 by joolswills