RetroPie/RetroPie-Setup #1128: Both PS3 controllers (BT) appear on the same port /dev/input/js0

Baklazan22, 2015-12-14:
When used one of the previous version of Retropie (and after using qtsxia to pair the fake ones) I was able to pair/connect both my PS3 controllers at once and play two-player games. Each of the controllers had its BT address as a part of name (e.g. "PLAYSTATION(R)3 Controller (00:26:5C:58:1C:72)"). One appeared as js0, the other as js0. Worked well.

After installing Retropie 3.21, it looks like both controllers (one after the other) appear as js0, and their BT address is something like 00:00:00, name in retroarch-joypads .cfg is just "PLAYSTATION(R)3 Controller", resulting in inability to pair both controllers at once. So, working well for one player (controller), but not working for 2 controllers config.

Any ideas how to pair them both?

gizmo98, 2015-12-17:
Two controller pairing should work. I have no problem pairing and playing with two original ps3 controllers.

Baklazan22, 2015-12-18:
As I wrote I have a fake ones. But this solved the problem (I can see my controllers with their MAC address again ;), maybe it could be included in the image by default:

git clone https://github.com/supertypo/qtsixa.git
cd qtsixa/sixad
make
sudo mkdir -p /var/lib/sixad/profiles
sudo checkinstall

gizmo98, 2015-12-18:
We already use parts of supertypos repo. Could be we do something different, but i have no fake controller to check the significant bit. The removed MAC address is not important.

Baklazan22, 2015-12-18:
Ok, thanks anyway. I have two "CECHZC2U", "BLUETOOTH DUALSCHOCHIII", EAN 711719801988 controllers from Ebay.

Just to make it crystal clear for others:
- install image (3.2.1) on SD card
- run setup script (sudo RetroPie-Setup/retropie_setup.sh)
- select "3  Setup / Configuration 9 (to be used post install)
- select "310 Install/Pair PS3 controller"
- select "1 /Install/Pair PS3 controller"
- wait until the installation finishes.

At this point an original PS3 controller should work using the "Plug/Unplug the USB cable and Press PS button" connection procedure. My fake ones didn't. Therefore run these following extra commands:

git clone https://github.com/supertypo/qtsixa.git
cd qtsixa/sixad
make
sudo mkdir -p /var/lib/sixad/profiles
sudo checkinstall
sudo reboot - I prefer it but not necessary

After this is completed the "Plug/Unplug the USB cable and Press PS button" connection procedure should work even for (my) fake ones. Connect the first controller and check it's working. Then repeat the the procedure to connect the second controller. Reboot again to be sure everything's working. Press PS button on both controllers and they should connect and work ok.

gizmo98, 2015-12-18:
Sorry. I have forgotten something ;-)
If you update RetroPie-Setup there is a new option:
```
select "2 /Install/Pair PS3 controller (clone support)"
```
Only this option has changes from supertypo.

Baklazan22, 2015-12-18:
I remember I saw it somewhere ;). Anyway, with that option the behavior described in my first post happened. I was able to somehow (not consistently) to connect first or second controller individially but not both at the same time. Steps above work 100%, at least for my setup.

The next challenge is making AirPlay work in KODI (installed from Experimental packages). This is the only feature I miss from OSMC. HDMI-CEC works fine, fusion tvaddons too, I can complile libtorrent-rasterbar + deluge for torrents and I have a perfect setup (+ LAMP and Arduino stuff ;).

Keep the work on Retropie Pie, it's a fantastic piece of work and effort, thanks for that.

gizmo98, 2015-12-19:
Could you update RetroPie-Setup again and use the "clone support" option? I added some hidden code from supertypo's repo.

Baklazan22, 2015-12-19:
The good thing: I can consistently connect the first or second controller after reboot or sixad restart - improvement from previous state when I was lucky to connect them I guess once after installation and later no luck whatsoever.

The bad thing: I still can't connect them both.

tried:
sixad --stop
sixad --start
-> I can connect one controller. I am not able to connect the second one unless I stop/start sixad again. Then I can connect the second one, but the first one get disconnected.

pi@retropie ~ $ sixad --stop
pi@retropie ~ $ sudo sixad --start
D-Bus setup failed: Name already in use
sixad-bin[2740]: started
sixad-bin[2740]: sixad started, press the PS button now
sixad-bin[2740]: Done
pi@retropie ~ $ sixad-sixaxis[2743]: started
sixad-sixaxis[2743]: Connected 'PLAYSTATION(R)3 Controller (00:26:5C:00:A7:EF)' [Battery 05]

* first one connected, Emulationstation sees it as Playstation(R)3 Controller (no MAC address in name)  *
* long wait for connecting the second controller (plug/unplug/PS button) => no luck *

^C

*** trying the second controller ***

pi@retropie ~ $ sixad --stop
pi@retropie ~ $ sudo sixad --start
D-Bus setup failed: Name already in use
sixad-bin[2816]: started
sixad-bin[2816]: sixad started, press the PS button now
sixad-bin[2816]: Done
sixad-sixaxis[2820]: started
pi@retropie ~ $ sixad-sixaxis[2820]: Connected 'PLAYSTATION(R)3 Controller (00:26:5C:58:1C:72)' [Battery 05]

* second one connected, Emulation station sees it as Playstation(R)3 Controller (no MAC address in name)  *
* first one disconnected  *

After the reboot the last connected controller will get connected after powering it up. The only way how to connect the second one (alone) is restaring the sixad while NOT starting the last connected controller.

On the othe hand, this is how it works on my setup (using the install procedure desribed above):

pi@retropie ~ $ sixad --stop
pi@retropie ~ $ sudo sixad --start
sixad-bin[2613]: started
sixad-bin[2613]: sixad started, press the PS button now
sixad-sixaxis[2648]: started
sixad-sixaxis[2648]: Connected 'PLAYSTATION(R)3 Controller (00:26:5C:00:A7:EF)' [Battery 05]
sixad-sixaxis[2690]: started
sixad-sixaxis[2690]: Connected 'PLAYSTATION(R)3 Controller (00:26:5C:58:1C:72)' [Battery 05]

Emulation station sees them as sixad - also with MAC addresses in their name.

BTW, looks I found another gitch. I wasn't able to get NeoGeo running with the working controller. Spend lot of time with fba2x.cfg and all other configs, eventually made 1:1 configs copy from my working 2.7 version, still no luck. As a last resort I checked the pifba file itself and the size was different. After I have overwritten the 3.2.1 version of fba2x (3452960 bytes) with the 2.7 version (3448848 bytes) the magic happened and controllers started woking.

samart45, 2015-12-25:
I use 2 ps3 clone controllers (gasia).
I select 2 Install/Pair PS3 controller.
I can connect only first controller. maybe service sixad terminate.

I try to recompile source from menu 1.

git clone https://github.com/RetroPie/sixad

cd sixad
make GASIA_GAMEPAD_HACKS=1 DEVICE_SHORT_NAME=1
sudo make install

reboot
Try to pair both controllers.
It's work.

Maybe gasia branch has some bug for second controller.
Please recheck it again. Thank you so much.

gizmo98, 2015-12-25:
I was hoping not doing this way. If you compile master branch with GASIA_GAMEPAD_HACKS=1 you cannot use original controllers at the same time. On the other hand branch gasia seems to work with only one gasia controller.

mspacemedia, 2016-01-21:
Hi just to confirm that in 3.4 the same issue occurs when trying to connect two ShanWan clones. Initial install and 'install/pair ps3 controllers (clones)' fails to connect two controllers at once.

Installing the supertypo repo of sixad fixes the issues and emulationstation sees both controllers fine after a reboot.

Would it be possible to switch to the supertypo repo for future releases?

Regards,

M

gizmo98, 2016-01-31:
@Baklazan22  @mspacemedia 
Please update RetroPie-Setup. There should be a install option for shanwan clones now.

mspacemedia, 2016-01-31:
Excellent @gizmo98 I'll give it a go soon, OT but screen blanking onload would also be a great option at some point ;)

gizmo98, 2016-02-20:
I will close this for now. My Shanwan controller is working and users report working gasia controllers as well.

closed 2016-02-20 by gizmo98

awd86, 2017-06-10:
@Baklazan22 your instructions worked really well, I only wish this thread had been easier to find.

RafaPolit, 2018-03-31:
I have successfully installed both controllers with the new options.

Still having issues:
- If I use gaia, they connect via bluetooth, but both share the same number in the controls (1)
- If I use Shanwan, each connect correctly and each show a different number (1) and (2).  But when playing, one of the controls behaves erratically and some buttons presses go through while others don't.

Any idea what could be happening?

Thanks.  Best regards,
Rafa.

Baklazan22, 2018-03-31:
@RafaPolit  Have no idea. I still need to use https://github.com/supertypo/qtsixa.git if I want to connect both of my controllers. I installed the last version of Retropie recently and none of the options in the menu allowed connecting of both controllers at the same time. 
So I used my 2 years old install procedure again and still works well.

RafaPolit, 2018-04-01:
I managed to connect both correctly by disabling the WiFi.

I use my RPi3 also for audio reproduction, and the audiophile community has reported issues with both Bluetooth and WiFi enabled.

I disabled the WiFi and everything worked as expected!!  Maybe it’s worth trying this for those having issues.

People report that using a Bluetooth dongle and keeping the WiFi is the best way to solve both issues.

For now I have to enable WiFi if I need to install anything, and then turn it of for playing.