RetroPie/RetroPie-Setup #1130: Play games from USB [Feature Request]

donmega, 2015-12-15:
Wouldn't be great to play games from a USB stick within the menu? That would be great :)

Keep up the good work!

joolswills, 2015-12-15:
You can do this by mounting the USB over ~/RetroPie/ or so. Something like http://electricgardener.net/2014/04/03/retropie-moving-games-usb-stick/ 

Might add something more automatic in the future, but for now, it's a manual process

craigerskine, 2016-01-15:
I've been playing my games from USB for a long while now because I was tired of having to re-upload my Roms every time a new RP image was released. I like to start fresh so as not to introduce legacy setting issues. Playing from USB also allows me to rename the systems folders to something more scoped so all brands are together and easy to find/read. The only real pain is having to edit the es_systems.cfg every time I update/install a new RP.

I'd really like to have a way to disable systems and choose rom paths for each system via some sort of GUI, preferably in ES. That would be ideal IMO.

zgoda, 2016-02-17:
Scraper does not work with roms directory on usb, either with system path changed in es_systems.cfg or with /media/usb symlinked to ~/RetroPie/roms (and dmask=0000 in /etc/usbmount/usbmount.conf to get rwxrwxrwx on system directories). Both situations lead to "No populated rom folders found".

joolswills, 2016-02-17:
@craigerskine mount the usb directly over ~/RetroPie/roms or somewhere else and bind mount it - saves editing paths. Alternatively you could symlink rom folders. You should avoid editing the sytems config if you have to - but if you want to manually change it - copy it to ~/.emulationstation and it won't be overwritten.

joolswills, 2016-02-17:
@zgoda you need to mount your usb so it is writable by the pi user - use the forum for help on this.

joolswills, 2016-02-17:
aah sorry you mean the experimental scraper ? probably wont work with symlinks - bind mount the usb or so for now.

joolswills, 2016-02-17:
http://backdrift.org/how-to-use-bind-mounts-in-linux

zgoda, 2016-02-17:
It is:
```
/dev/sda1 on /media/usb0 type vfat (rw,nodev,noexec,noatime,nodiratime,sync,uid=1000,gid=1000,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=ascii,shortname=mixed,flush,errors=remount-ro)
```

joolswills, 2016-02-17:
ok - was a simple fix to make the scraper module see the roms when symlinked - please update retropie-setup and try again (do a full binary install if coming from an older retropie release)

zgoda, 2016-02-17:
Works like charm, thank you.

joolswills, 2017-05-31:
We now include a simple way with our usbromservice to automatically mount usb drives which is documented on the wiki/retropie docs. Closing this.

closed 2017-05-31 by joolswills