RetroPie/RetroPie-Setup #1132: Retro Achievements

kryptalivian, 2015-12-15:
Can we get Retro Achievements implemented from retroachievements.org. I see Retroarch is implementing this into builds.

Thanks!

robertybob, 2015-12-18:
Lakka has this, so I wondered whether the version of RetroArch that RP uses has this feature already, hidden away?

HerbFargus, 2015-12-18:
It will eventually be incorporated into retroarch once retroarch sorts out their 1.3 release and its deemed stable enough to be compiled on retropie.

joolswills, 2015-12-22:
It is compiled in but untested. Because you can't enter the username/password from within RGUI it will need manual configuration.

kryptalivian, 2015-12-22:
So we just need to enter it into the Retroarch.cfg file?

joolswills, 2015-12-22:
try setting these in /opt/retropie/configs/all/retroarch.cfg

```
cheevos_username = "yourusername"
cheevos_password = "yuourpassword"
cheevos_enable = true
```

kryptalivian, 2015-12-22:
I'll try it tonight and let you know if it was successful. Thanks for getting that all setup! :-)

zerojay, 2015-12-31:
Since OP hasn't commented, this appears to not be working yet.

kryptalivian, 2015-12-31:
Test it with Chrono Trigger and confirmed working! :-D

zerojay, 2016-01-01:
Ah, you're right. The issue here is that we don't have a core that supports achievements on NES which is an upstream issue.

joolswills, 2016-01-09:
Can this be closed now ?

kryptalivian, 2016-01-09:
Yes, but can we get a compatible nes emulator?

zerojay, 2016-01-10:
That's going to have to come from upstream, isn't it?

I would say this should be closed.

gizmo98, 2016-01-10:
Seems to be QuickNES supports achievements:
https://pitendo.wordpress.com/2016/01/07/adding-achievements-to-nes/

We have no QuickNES module atm.

closed 2016-01-13 by kryptalivian

zerojay, 2016-02-16:
I've submitted a pull request for the QuickNES module specifically to support achievements for the NES platform: https://github.com/RetroPie/RetroPie-Setup/pull/1270