RetroPie/RetroPie-Setup #1136: File System 'Expanded' automatically

robertybob, 2015-12-18:
Unless there's a really good reason I'm missing, I can't see why users should have to do this themselves when setting up RetroPie for the first time. Some people forget and then wonder why there's no room for ROMs. This would eliminate that problem (?). 

Is there a way of the file system being expanded automatically, the very first time a user boots up a fresh copy of RetroPie?

biscuits99, 2015-12-19:
I guess there is a way to do it automatically as Recalbox does it,
Although what if someone wants to get RetroPie running, make a range of config changes then back up the image without having to shrink it again? Or is that an edge case?

robertybob, 2015-12-19:
Hmm yeah I didn't think about that situation, although I would like to imagine it's only a small number of users doing that. I just thought it's a small way to make RetroPie even more newbie friendly.
What are your thoughts @joolswills @herbfargus ?

joolswills, 2015-12-20:
My thoughts are that there are more important things currently, and I don't like forcing it really (what if they wanted to make another partition on the SD card once they boot ?) - I also think that doing things like repartitioning/growing filesystems should be things that are done with the users knowledge/confirmation amnd Raspbian doesn't do it automatically either.

The "some people forget" thing doesn't sound like a good reason to me. Don't forget ? :) I think it's good that users are encouraged to read the documentation too.

joolswills, 2015-12-20:
Short answer. I don't want to change it.

closed 2015-12-20 by robertybob