RetroPie/RetroPie-Setup #1137: Problem with Thrustmaster Run'N' Drive Wireless PS3 Controller

NosovDN, 2015-12-20:
Hello everybody,
I have RPi2 with retropie 3.2.1 and wireless controller. Every boot input configuration dialog is appeared. I walk through it every time correct. But comething goes wrong:

1. after exiting from game using hotkeys (Select+Start) joypad doesn't work. Restart app helps only.
2.  input configuration dialog appear EVERY boot/launch app. 

There is some TS information
I have the following messages in console:
```
input config finish command:
Input type is 'joystick'.
Configuring 'emulationstation'
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3'])'
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='pagedown'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='pagedown'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='start'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='start'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='pageup'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='pageup'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='up'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='up'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='a'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='a'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='b'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='b'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='down'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='down'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='right'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='right'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='select'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='select'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid predicate: count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='left'])
compilation error: element with-param
XSLT-with-param: Failed to compile select expression 'count(/inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']/input[@name='left'])'
Invalid predicate: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Invalid expression: /inputList/inputConfig[@deviceName='Thrustmaster Run'N' Drive Wireless PS3']
Configuring 'retroarch'
==============================================
```
My /home/pi/.emulationstation/es_input.cfg is following
```
<?xml version="1.0"?>
<inputList>
  <inputAction type="onfinish">
    <command>/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh</command>
  </inputAction>
  <inputConfig type="joystick" deviceName="Thrustmaster Run'N' Drive Wireless PS3"/>
  <inputConfig type="joystick" deviceName="Thrustmaster Run'N' Drive Wireless PS3"/>
  <inputConfig type="joystick" deviceName="Thrustmaster Run'N' Drive Wireless PS3"/>
  <inputConfig type="joystick" deviceName="Thrustmaster Run'N' Drive Wireless PS3"/>
</inputList>

```
My /opt/retropie/configs/all/retroarch-joypads/ThrustmasterRun\'N\'DriveWirelessPS3.cfg
```
input_device = "Thrustmaster Run'N' Drive Wireless PS3"
input_driver = "udev"
input_r_y_plus_axis = "+3"
input_l3_btn = "10"
input_r_x_minus_axis = "-2"
input_r_btn = "7"
input_save_state_btn = "7"
input_l2_btn = "4"
input_start_btn = "9"
input_exit_emulator_btn = "9"
input_l_y_minus_axis = "-1"
input_l_btn = "6"
input_load_state_btn = "6"
input_up_btn = "h0up"
input_r_y_minus_axis = "-3"
input_a_btn = "1"
input_b_btn = "2"
input_reset_btn = "2"
input_down_btn = "h0down"
input_r2_btn = "5"
input_l_x_plus_axis = "+0"
input_l_y_plus_axis = "+1"
input_r3_btn = "11"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_x_btn = "0"
input_menu_toggle_btn = "0"
input_select_btn = "8"
input_enable_hotkey_btn = "8"
input_l_x_minus_axis = "-0"
input_y_btn = "3"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_r_x_plus_axis = "+2"
```

I think there is a problem with an apostrophes in the name - Thrustmaster Run'N' Drive Wireless PS3.
But how can I solve it. There were no such problems in previous releases - I sould configure my controllers by myself. And everything worked fine. Could You be so kind to help me to find workaround?

joolswills, 2015-12-20:
There were some bugfixes for apostrophes. I thought I did them before the 3.2.1 release though - please can you update the retropie-setup script, and then reinstall emulationtation from binary from menu 5

joolswills, 2015-12-20:
(and then retest)

joolswills, 2015-12-20:
Aah no this was fixed after the 3.2.1 release - so the above steps should sort it.

NosovDN, 2015-12-21:
Great thank's. I will try it in a few hours.

NosovDN, 2015-12-21:
I've done as You said - everything works fine. Thank's again.

joolswills, 2015-12-21:
Cheers.

closed 2015-12-21 by joolswills