RetroPie/RetroPie-Setup #1138: Xbox 360 and One controllers can not be configured using "Configure Input"

SirJonthe, 2015-12-22:
Both the Xbox 360 and Xbox One controller triggers can not be set up properly using Main Menu -> Configure Input.

Pressing a trigger button will do one of the following 3 things:

1) When pressing a trigger without bottoming it out the button setup will simply be skipped (identical behavior to holding any button to skip)
2) When pressing a trigger and bottoming it out the button setup will assign the appropriate Axis X+ to the selected button, Axis X- to the subsequent button, and skip the button setup following the previous two (identical behavior to holding any button to skip).
3) When pressing a trigger and bottom it out the button setup will assign the appropriate Axis X+ to the selected button and skip the subsequent button setup (identical behavior to holding any button to skip). 

Note the following:
* xboxdrv is installed, system has been rebooted.
* System is otherwise a completely fresh install of Retropie 3.0 (public release; not RC, beta, or alpha).
* Main Menu -> Configure Input detects 360 controller as "MICROSOFT X-BOX 360 PAD" and One controller as "MICROSOFT X-BOX ONE PAD".

robertybob, 2015-12-22:
This may be your best bet :) 
http://blog.petrockblock.com/forums/forum/retropie-project-forum/controller-configuration-in-retropie/

joolswills, 2015-12-22:
Sounds like the xboxdrv isn't running. A fair amount has changed since 3.0 also.

Also when configuring you can move up after configuring all the controls when you are on "OK" to correct the missing button configuration.

You could start from a 3.3 image which has a newer xboxdrv also (however there is a bug that requires updating of retropie-setup, then disabling/re-enabling xboxdrv)

You might do better to use the forum for help on setting up controllers - there has been previous discussion of correcting configuration when triggers don't work.

joolswills, 2015-12-22:
also please detail how you installed "xboxdrv" and post the contents or /etc/rc.local

SirJonthe, 2015-12-22:
@joolswills I had no idea I could move up to correct input. Good to know. I installed xboxdrv via SSH and "sudo apt-get install xboxdrv". I am unable to post contents of /etc/rc.local as I have already scrapped my install of Retropie unfortunately. Sorry.

joolswills, 2015-12-22:
installing xboxdrv like that will not start it up after a reboot - it will just install the driver package/ There is an option in retropie-setup for enabling xboxdrv, or you need to start it up manually yourself as detailed here

https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller

Using our premade configuration added to /etc/rc.local via our configuration script is the easiest way.

joolswills, 2015-12-22:
Closing as I believe this is a configuration problem, and if you are no longer using RetroPie there is no need for any further support.

closed 2015-12-22 by joolswills