RetroPie/RetroPie-Setup #1139: [lr-reicast] testing

nemo93, 2015-12-22:
I just saw a new "libretro" repo has been created here https://github.com/libretro/reicast-emulator. Needs further testing. Will keep this thread updated. Needs also comparison with the "non lr" reicast version from Retropie.

robertybob, 2015-12-22:
Great news!

robertybob, 2016-03-04:
Slightly off-topic, but does this have any impact on the non-reicast version that RetroPie uses? Will different flags be needed when compiling? https://github.com/reicast/reicast-emulator/pull/832

joolswills, 2016-03-04:
what do you mean by "impact" ?

joolswills, 2016-03-04:
We will need to update our module when our RPI branch is updated, and update for the X11 build to use SDL2.

joolswills, 2016-03-04:
This ticket is unrelated - please use foe forum for discussion etc.

joolswills, 2016-03-04:
(or open a new ticket if you have an issue with something related to development)

gizmo98, 2016-03-04:
Lr-reicast cannot be build atm.

gizmo98, 2016-03-04:
And sdl2 support would be nice if input setup is easier.

joolswills, 2016-05-10:
we include a lr-reicast module now (thanks @gizmo98) so I'm going to close this - doesn't work on RPI, but it can be tested again in the future.

closed 2016-05-10 by joolswills