RetroPie/RetroPie-Setup #1140: USB controller converter gets added twice.

phaseIV, 2015-12-23:
Hi and congrats on a new release! I have this issue where my USB controller converter (it's a TwinUSBJoystick, for connecting two ps1 controllers) is registered twice when entering a game.
After a game is started the yellow line at the bottom reads; 'TwinUSBJostick configured in port #0', 'TwinUSBJostick configured in port #1' and again 'TwinUSBJostick configured in port #0', 'TwinUSBJostick configured in port #1'. So in the end the RGui menu gives me four controllers to choose the input from.
It is caused by this setting in retroarch.cfg;
# Input device driver. (Valid: linuxraw, sdl, dinput)
input_joypad_driver = sdl2
I tried changing the driver in retroarch-joypads/TwinUSBJoystick.cfg from 'udev' to 'sdl2' but the issue remains.

gizmo98, 2015-12-23:
Change `input_joypad_driver = sdl2` into `input_joypad_driver = udev`.

phaseIV, 2015-12-27:
Well, yeah, that’ll fix the issue but I prefer to use sdl2 since it treats analog devices much nicer.

I already used sdl2 with older RetroPie versions (and experienced the issue described above) but since sdl2 became the default driver (in 3.3) I decided to file it as an issue.

joolswills, 2015-12-27:
I would report it upstream to RetroArch then.

phaseIV, 2015-12-27:
Okay, I'll move over to RetroArch then, thanks. With an issue like this it's hard to specify where it happens and what territory it belongs to. Close this issue if you like.

joolswills, 2015-12-27:
Understood - thanks.

I wonder if we should change the default back to udev - there are other issues with the sdl2 driver it seems in regards to matching autoconfig against device name. I will do some further teting.

joolswills, 2016-01-04:
The driver has now been changed back to udev.

closed 2016-01-20 by joolswills