RetroPie/RetroPie-Setup #1143: USB Rom Service Issue (v3.3 Jessie)

LodanZark, 2015-12-24:
Using the USB Rom Service over Retropie 3.3 Jessie I've got this issue:

after insert the usb on rpi

the emulationstation loading screen starts flashing and shows up the command line thiserrors:
,,,
terminate called after throwing an instance 'boost::filesystem::filesystem_error'

what(): boost::filenamesystem::directory::iterator::construct: Permission denied "home/pi/RetroPie/roms/megadrive/"

/usr/bin/emulatuionstation line 20: 366 Aborted      "$es_bin $@"
,,,
and  the abortion number starts increase each time

joolswills, 2015-12-24:
you have a permissions issue - possibly due to the permissions on the usb stick when the files were copied. Please use the option in retropie-setup to fix up rom permissions and it should be ok.

joolswills, 2015-12-24:
How did you format the usb stick ?

LodanZark, 2015-12-24:
I had formatted the usb stick with fat32 format, I did the same procedure as I have done in previews versions, I cannot figured out what I did wrong

joolswills, 2015-12-24:
Did you take it out too early ? After rsyncing the files, it then changes the ownership. Maybe something like that but I will test it works as it should on Jessie. Did my fix above work ?

LodanZark, 2015-12-24:
I just give a try, you're right, after reset the rom folder permissions it fixed the issue. thanks ;)

closed 2015-12-24 by joolswills

sneakypete76, 2017-10-10:
How does one change the permission settings in retropie please? 

A friend passed me a Pi to configure for him. 
Rather than mess with the SD card directly, I figured it would be safer/easier to use a USB thumbdrive to add the folders, after adding the desired files on windows (drag and drop), I safely ejected and placed it back in the Pi. After the light stopped flashing, I restarted Emulation Station from the menu. Everything was working until I powered down the unit and restarted. Now it wont boot into emulation station and complains of  ""erminate called after throwing an instance of" 'boost::filesystem_error' "" no matter whether the USB drive is plugged in or not.

How can I fix it please?

Thank you

joolswills, 2017-10-10:
You need to use the RetroPie forum for support.

sneakypete76, 2017-10-10:
Thanks. The forum says:
"Can't change emulator for ROM, permissions"
Solution: in retropie-setup -> configuration there's a 'reset permissions' (or similar) tool you should use.
[–]cuckfupertino[S] 2 points 2 months ago 
This seems to have resolved it. Thanks!

It doesnt explain how to do it. I have no idea how to get to the retropie setup. I'll keep digging. It's just that my search brought me here to someone with a solution but noone seems to be able to explain how its done.

joolswills, 2017-10-10:
Post to the retropie forum detailing your issue providing information about your set-up (it details this on registration) - this issue tracker is not for support.

sneakypete76, 2017-10-10:
OK thanks

sneakypete76, 2017-10-10:
Can they try to resolve the issue for future builds please or at least update the tutorial to reflect this issue and rectify it as part of the initial user set-up instead of leaving it hanging at a critical error at the endo of the tutorial. Thanks.

joolswills, 2017-10-10:
Please stop posting here. Ask on the forum.

FerrariGlenn, 2017-10-27:
How does one change the permission settings in retropie please?

joolswills, 2017-10-27:
Use the forum

sneakypete76, 2017-10-27:
Thanks. I've used the forum.

FerrariGlenn, 2017-10-27:
no you use the forum

sneakypete76, 2017-10-27:
Thanks. Like I say. I am using the forum.