RetroPie/RetroPie-Setup #1144: Overscan Disabled (empty config.txt)

rastolpho, 2015-12-24:
I noticed a change from 3.2.1 to 3.3 on my rPi2. Emulationstation had the black borders around the screen.

Looked into /boot/config.txt and added disable_overscan=1 but did not fix. Only removing the SDCard from the Pi, loading it up on Windows to open config.txt on Wordpad I was able to do so. Config.txt on Windows was also completely different than config.txt on /boot. Is that normal?

I also could not overclock and had to manually add the arm_freq, core_freq, sdram_freq and over_voltage lines into the config.txt (while viewing on Windows) to make it work. I believe this was brought up already and will be fixed.

joolswills, 2015-12-24:
Please see https://github.com/RetroPie/RetroPie-Setup/issues/1141

joolswills, 2015-12-24:
If you make the corrections in the other ticket, you will have the data in /boot again/

joolswills, 2015-12-24:
There is an updated image available now with the correction in place also (3.3.1)

closed 2015-12-24 by joolswills