RetroPie/RetroPie-Setup #1145: Error Lines when launching ROM

rastolpho, 2015-12-24:
Saw this happening on Wheezy 3.3 and also on 3.3.1:

When launching a ROM (tried different emulators such as SNES, SMS, ATARI 2600), when the screen goes dark to load there are 2 lines that show up that did not show on 3.2.1:

"Didn't read whole file: /sys/devices/system/cpu/present."
"Didn't read whole file: /sys/devices/system/cpu/possible."

The ROM launches just fine though and I am able to play with no problems.

objectchris, 2015-12-24:
This is a retroarch issue and might be fixed.

See https://github.com/libretro/RetroArch/issues/2574#issuecomment-167046166
Try rebuilding retroarch from source in the Retropie > Retropie-Setup > 'Install Individual Emulators' section. If it still doesn't work I guess update that issue?

joolswills, 2015-12-24:
Those errors can also just be ignored. It will not affect RetroArch.

joolswills, 2015-12-24:
Actually I will leave this open for now as a reference for others.

fastfalcon94, 2015-12-29:
I was seeing this as well. I did what objectchris recommended and it fixed it. It does seem like the linux prompt window hangs for a bit now. It says press any key within 5 seconds to keep it from launching emulstation again. Sometimes it takes 20+ seconds before it starts to launch it again. Sometimes it doesn't spit out the "press any key" message. Instead it ends on the ", to the extent permitted by applicable law." message and I have to press ctrl+c to get a prompt back. When I do the prompt is owned by a cpi user. This all started after updating and running the setup script last night. Before I was using the 3.3 version delivered with the flash image.

joolswills, 2015-12-29:
That delay is unrelated - ES saves all the config files on exit (unless quitting with f4), so it can take a while to restart if you have lots of roms.

fastfalcon94, 2015-12-29:
Ah. Well I also get the delay with F4. But that would explain it. I just rebuilt my system on a bigger 32gb microsd and copied over all my roms before upgrading.Thanks!

joolswills, 2016-01-30:
Closing as issue in OP is now sorted in RetroArch

closed 2016-01-30 by joolswills