RetroPie/RetroPie-Setup #1147: Dreamcast

roydenstorm, 2015-12-24:
Have the Dreamcast emulator working, it loads no problem, but I cant get passed the "set date/time" screen, I am able to change the date/time, so left/right and up/down work, I just cannot use "select" button. I've tried changing the config files for my controller(xbox 360 wireless) and anything else I could think of. Any and all help would be greatly appreciated.

p.s. I also just got the Intellivision emulator working though I am having some controller issues with it as well, I just haven't taken a look yet to see if I can fix it, but if anyone already knows what to do please let me know :)

Thanking you all in advance for your time and effort it will be greatly appreciated :)

joolswills, 2015-12-24:
Please use the forum for help - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2015-12-24 by joolswills

roydenstorm, 2015-12-24:
ive been trying to post to that forum but cannot seem to find anywhere that allows me to create anew post lol, really user friendly that site is lol

gizmo98, 2015-12-24:
Press A/B/X or Y. Don't press Select. I don't think you will get any help today. It's Christmas. No one should be on the forums now.

gizmo98, 2015-12-24:
Just for information. The dreamcast controller had no select button.

roydenstorm, 2015-12-24:
though I appreciate the response I don't think you're reading my problem correctly. The "select" button is on-screen, maybe  take a look at the dreamcast emulator and see what I'm talking about :)

roydenstorm, 2015-12-24:
and ill never get help on that forum because I cant even find the option to create a new post lol

InsecureSpike, 2015-12-24:
got the forum you want to post in, then scroll right to the bottom, that gives you the option to create a new topic......

gizmo98, 2015-12-24:
Narf. It states "select" but means press X Y or A B. ;-) 
If you press select reicast quiets.

gizmo98, 2015-12-24:
It just means "select this date" and does not mean "press select". Ok, it os confusing 😉

roydenstorm, 2015-12-24:
I tried every button on my controller and keyboard long before I came to this forum for help, I appreciate your trying but still stuck on that damn screen ;(

roydenstorm, 2015-12-24:
lol well pressing every single button on my controller and keyboard does absolutely nothing!!! so yes it is so confusing

joolswills, 2015-12-24:
Please use the forum as requested.