RetroPie/RetroPie-Setup #1153: Input Bluetooth Controller error

sklivvz, 2015-12-26:
RetroPie latest
8bitdo SFC30 (firmware 1.63)
Raspberry Pi 2
Dongle Bluetooth 4.0 ASUS

I am trying to pair my retropie with two [8BITDO SFC30](http://www.8bitdo.com/sfc30/) gamepads.

To do so, I start them in the appropriate mode (pressing START+B on one and START+RIGHT+B on the other). I then connect them with this script:

    sudo bluez-test-device remove $1
    sudo bluez-simple-agent hci0 $1 -c DisplayYesNo
    sudo bluez-test-device trusted $1 yes
    sudo bluez-test-input connect $1

I am confident the above steps are correct, but I get a `dbus` exception. For example:

    pi@retropie ~ $ sudo ./connect.sh 01:02:03:04:05:06 
    Release
    New device (/org/bluez/2270/hci0/dev_01_02_03_04_05_06)
    Traceback (most recent call last):
      File "/usr/bin/bluez-test-input", line 40, in <module>
        input.Connect()
      File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
        return self._proxy_method(*args, **keywords)
      File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
        **keywords)
      File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
        message, timeout)
    dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "Connect" with signature "" on interface "org.bluez.Input" doesn't exist

I strongly suspect it's a bug, but I don't know what it means.

joolswills, 2015-12-26:
Raspbian Wheezy or Jessie ?

It's not a RetroPie bug at least - you are pairing manually, which is not connected to the RetroPie-Setup script but the underlying OS/

you can try and pair them / connect using retropie-setup -> setup configuration -> Configure Bluetooth Devices

You will likely need to edit /etc/rc.local to connect automatically on boot through. This is detailed in step 3 on https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller (you shouldn't need to manually pair as our helper script may work to do that for you).

joolswills, 2015-12-26:
I have updated step 3 in the wiki, as it mentioned editing /etc/init.d/rc.local but /etc/rc.local is a better place.

sklivvz, 2015-12-26:
Connecting via the script does exactly the same:

    = = = = = = = = = = = = = = = = = = = = =
    Configuring 'bluetooth' : Configure Bluetooth Devices
    = = = = = = = = = = = = = = = = = = = = =
    
    Traceback (most recent call last):
      File "/usr/bin/bluez-test-input", line 40, in <module>
        input.Connect()
      File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
        return self._proxy_method(*args, **keywords)
      File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
        **keywords)
      File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
        message, timeout)
    dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "Connect" with signature "" on interface "org.bluez.Input" doesn't exist

There is also another bug as the configure script doesn't catch the above error and reports success.

joolswills, 2015-12-26:
Are you using Raspbian Wheezy or Jessie ?

Regarding the "module" - It doesn't pick up every error. Unfortunately the python helper scripts that come with the bluetooth package don't allow easy automation.

Problem may be bluetooth dongle then, or something else. Did you follow all the instructions on the wiki - as that is written by someone with the similar controllers and they do work. Firmware up to date ?

You could try and get some help from the vendor perhaps. Note that the bluetooth stack has nothing to do with RetroPie-Setup, and is part of the underlying Raspbian OS.

sklivvz, 2015-12-26:
More system info:

    pi@retropie ~/RetroPie-Setup $ uname -a
    Linux retropie 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux

and 

    pi@retropie ~/RetroPie-Setup $ cat /etc/debian_version
    7.8

joolswills, 2015-12-26:
You might want to try seeing if anyone on the forum has successfully paired the devices - and you could try the raspberrypi forum also. 

Also to see if anyone else on the RPI has used a similar dongle (lsusb)

http://blog.petrockblock.com/forums/forum/retropie-project-forum/controller-configuration-in-retropie/

joolswills, 2015-12-26:
Also make sure that "/etc/bluetooth/main.conf" doesn't have "DisablePlugins" enabled with the input plugin. If you google the error you will find other references to it on raspberrypi forum. Other things mentioned were incompatible bluetooth dongle

sklivvz, 2015-12-26:
Yes, I follow the steps of the guide and they break at the last command of step 2.

The problem is that I am trying to connect *two* devices and not *one*. In order to do so, they need to be started in "Mode 2" (Bluetooth keyboard) and not "Mode 1" (Joystick).

Using them in "Mode 1" only allows me to use one at a time -- I can pair and connect the devices, but only one at a time.

joolswills, 2015-12-26:
Would have been useful to mention that - Did you try without the "-c DisplayYesNo" parameter in "mode 2".

You should be able to pair two in "mode 1". You may need to do "hciconfig hci0 up piscan" before pairing the second.

joolswills, 2015-12-26:
Can't really advise further. I suggest asking the vendor for help or trying another bluetooth dongle.

joolswills, 2015-12-26:
Missed the last line. So one pairs - try the hciconfig line to pair the second then - might work then.

sklivvz, 2015-12-26:
`sudo hciconfig hci0 up piscan` seems to have done the trick.

There are different problems now, but I think I just need to configure emulation station.

Thanks.

closed 2015-12-26 by sklivvz