RetroPie/RetroPie-Setup #1154: PS4 Bluetooth with RetroPie

mrpangaio, 2015-12-27:
Is it possible to use a PS4 controller via Bluetooth in RetroPie?  If so, does anyone have instructions on setting up a PS4 controller via Bluetooth in RetroPie?   I successfully installed and used a PS3 controller both wired and via Bluetooth but the PS4 did not work using the same method.  Any help would be appreciated!

craigerskine, 2015-12-29:
FYI: I think you are supposed to ask these types of questions on the Retropie site.

I created a post about this awhile back... I just updated with a process that works for me every time I do a fresh RP install and want to pair my DS4 controller:
http://blog.petrockblock.com/forums/topic/dual-shock-4/#post-112640

joolswills, 2015-12-29:
Thanks - @mrpangaio please use the RetroPie forum for help. This is described at the top of the https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md document which is shown before opening an issue.

closed 2015-12-29 by joolswills