RetroPie/RetroPie-Setup #1155: Controller configuration fails

nschloe, 2015-12-27:
As of recently, [people are having problems getting their controllers configured](http://blog.petrockblock.com/forums/topic/controller-is-always-not-configured/). The controller works nicely in EmulationStation, but when starting any game, a yellow
```
controler xyz not configured
```
will appear on the screen and controls don't work. I can reproduce this on Jessie 3.3, others have the issue on Wheezy.

joolswills, 2015-12-27:
Looking at the thread it looks specific to certain controllers - is that correct ? 

I need some files to aid in debugging this as I don't own the controller in question.

Please can you post a copy of (after configuring controls in ES)

~/.emulationstation/es_temporaryinput.cfg
/
any files in /opt/retropie/configs/all/retroarch-joypads/

and after configuring controls in ES ssh in and run

```
/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
```

and post the output. thanks.

Javison, 2015-12-27:
Hi,

Seems that the "Automatic Configuration > Xbox / Xbox 360 gamepad driver" is not working properly on version 3.3. 
I cant find the folder "/etc/rc.local" or "/opt/retropie/supplementary/xboxdrv" after use the Automatic configuration and reboot.

output /opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh:
Input type is 'joystick'.
Configuring 'emulationstation'
Configuring 'mupen64plus'
Configuring 'reicast'
Configuring 'retroarch'

[es_temporaryinput.cfg.txt](https://github.com/RetroPie/RetroPie-Setup/files/72719/es_temporaryinput.cfg.txt)
[MicrosoftX-Box360pad.cfg.txt](https://github.com/RetroPie/RetroPie-Setup/files/72720/MicrosoftX-Box360pad.cfg.txt)

Thanks

joolswills, 2015-12-27:
@Javison please open a new issue for that - this thread is not related to the xbox driver

joolswills, 2015-12-27:
@Javison /etc/rc.local is a file and should be present - did you upgrade to 3.3.1 or use the image ? Update retropie-setup and then disable/re-eable the xbox 360 driver from retropie-setup.

nschloe, 2015-12-27:
@joolswills 

* The file `~/.emulationstation/es_temporaryinput.cfg ` is empty: 
```
pi@retropie:~ $ cat ~/.emulationstation/es_temporaryinput.cfg 
pi@retropie:~ $
```

* So is `/opt/retropie/configs/all/retroarch-joypads/LogitechGamepadF310.cfg`
```
pi@retropie:~ $ cat /opt/retropie/configs/all/retroarch-joypads/LogitechGamepadF310.cfg 
pi@retropie:~ $
```

* 
```
$ /opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
/home/pi/.emulationstation/es_temporaryinput.cfg:1.1: Document is empty

^
/home/pi/.emulationstation/es_temporaryinput.cfg:1.1: Start tag expected, '<' not found

^
/home/pi/.emulationstation/es_temporaryinput.cfg:1.1: Document is empty

^
/home/pi/.emulationstation/es_temporaryinput.cfg:1.1: Start tag expected, '<' not found

^
/home/pi/.emulationstation/es_temporaryinput.cfg:1.1: Document is empty

^
/home/pi/.emulationstation/es_temporaryinput.cfg:1.1: Start tag expected, '<' not found

^
Input type is ''.
Configuring 'emulationstation'
Configuring 'mupen64plus'
Configuring 'reicast'
Configuring 'retroarch'
```

joolswills, 2015-12-27:
odd - es_temporaryinput.cfg should be generated by emulationstation.

Please can you post the contents of ~/.emulationstation/es_log.txt (after configuring the device)

please can you also remove /home/pi/.emulationstation/es_temporaryinput.cfg and then re-configure the controller inide ES and see if it is generated

nschloe, 2015-12-27:
```
$ cat ~/.emulationstation/es_log.txt
lvl2: 	EmulationStation - v2.0.1a, built Oct 10 2015 - 03:00:02
lvl2: 	Creating surface...
lvl2: 	Created window successfully.
lvl2: 	Added known joystick Logitech Gamepad F310 (instance ID: 0, device index: 0)
lvl2: 	Checking available OpenGL extensions...
lvl2: 	 ARB_texture_non_power_of_two: MISSING
lvl2: 	Loading system config file /etc/emulationstation/es_systems.cfg...
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/amstradcpc/gamelist.xml"...
lvl1: 	System "amstradcpc" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari2600/gamelist.xml"...
lvl1: 	System "atari2600" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari5200/gamelist.xml"...
lvl1: 	System "atari5200" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari7800/gamelist.xml"...
lvl1: 	System "atari7800" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari800/gamelist.xml"...
lvl1: 	System "atari800" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atarilynx/gamelist.xml"...
lvl1: 	System "atarilynx" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atarist/gamelist.xml"...
lvl1: 	System "atarist" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/c64/gamelist.xml"...
lvl1: 	System "c64" has no games! Ignoring it.
lvl1: 	  Unknown platform for system "coco" (platform "coco" from list "coco")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/coco/gamelist.xml"...
lvl1: 	System "coco" has no games! Ignoring it.
lvl1: 	  Unknown platform for system "dragon32" (platform "dragon32" from list "dragon32")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/dragon32/gamelist.xml"...
lvl1: 	System "dragon32" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/dreamcast/gamelist.xml"...
lvl1: 	System "dreamcast" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/fba/gamelist.xml"...
lvl1: 	System "fba" has no games! Ignoring it.
lvl1: 	  Unknown platform for system "fds" (platform "fds" from list "fds")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/fds/gamelist.xml"...
lvl1: 	System "fds" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gamegear/gamelist.xml"...
lvl1: 	System "gamegear" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gb/gamelist.xml"...
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gba/gamelist.xml"...
lvl1: 	System "gba" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gbc/gamelist.xml"...
lvl1: 	System "gbc" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/intellivision/gamelist.xml"...
lvl1: 	System "intellivision" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mame-advmame/gamelist.xml"...
lvl1: 	System "mame-advmame" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mame-libretro/gamelist.xml"...
lvl1: 	System "mame-libretro" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mame-mame4all/gamelist.xml"...
lvl1: 	System "mame-mame4all" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mastersystem/gamelist.xml"...
lvl1: 	System "mastersystem" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/megadrive/gamelist.xml"...
lvl1: 	System "megadrive" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/msx/gamelist.xml"...
lvl1: 	System "msx" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/n64/gamelist.xml"...
lvl1: 	System "n64" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/neogeo/gamelist.xml"...
lvl1: 	System "neogeo" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/nes/gamelist.xml"...
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/ngp/gamelist.xml"...
lvl1: 	System "ngp" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/pcengine/gamelist.xml"...
lvl1: 	System "pcengine" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/psp/gamelist.xml"...
lvl1: 	System "psp" has no games! Ignoring it.
[... 5 lines cut ...]
lvl1: 	System "sega32x" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/segacd/gamelist.xml"...
lvl1: 	System "segacd" has no games! Ignoring it.
lvl1: 	  Unknown platform for system "sg-1000" (platform "sg-1000" from list "sg-1000")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/sg-1000/gamelist.xml"...
lvl1: 	System "sg-1000" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/snes/gamelist.xml"...
lvl1: 	  Unknown platform for system "vectrex" (platform "vectrex" from list "vectrex")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/vectrex/gamelist.xml"...
lvl1: 	System "vectrex" has no games! Ignoring it.
[... cut ...]
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/zxspectrum/gamelist.xml"...
lvl1: 	System "zxspectrum" has no games! Ignoring it.
lvl2: 	 req sound [basic.launch]
lvl2: 	   (missing)
```

> please can you also remove /home/pi/.emulationstation/es_temporaryinput.cfg and then re-configure the controller inide ES and see if it is generated

After having removed the file, the pad continues to work in ES, and continues to not work in games. How can I "re-configure the controller inside ES"?

joolswills, 2015-12-27:
press your button configured as start and go to configure input. I need to see the log files and /home/pi/.emulationstation/es_temporaryinput.cfg after doing this.

joolswills, 2015-12-27:
and please post the contents of /home/pi/.emulationstation/es_input.cfg

nschloe, 2015-12-27:
> press your button configured as start and go to configure input.

Got it.

> I need to see the log files 
```
cat ~/.emulationstation/es_log.txt
lvl2: 	EmulationStation - v2.0.1a, built Oct 10 2015 - 03:00:02
lvl2: 	Creating surface...
lvl2: 	Created window successfully.
lvl2: 	Added known joystick Logitech Gamepad F310 (instance ID: 0, device index: 0)
lvl2: 	Checking available OpenGL extensions...
lvl2: 	 ARB_texture_non_power_of_two: MISSING
lvl2: 	Loading system config file /etc/emulationstation/es_systems.cfg...
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/amstradcpc/gamelist.xml"...
lvl1: 	System "amstradcpc" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari2600/gamelist.xml"...
lvl1: 	System "atari2600" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari5200/gamelist.xml"...
lvl1: 	System "atari5200" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari7800/gamelist.xml"...
lvl1: 	System "atari7800" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atari800/gamelist.xml"...
lvl1: 	System "atari800" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atarilynx/gamelist.xml"...
lvl1: 	System "atarilynx" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/atarist/gamelist.xml"...
lvl1: 	System "atarist" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/c64/gamelist.xml"...
lvl1: 	System "c64" has no games! Ignoring it.
lvl1: 	  Unknown platform for system "coco" (platform "coco" from list "coco")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/coco/gamelist.xml"...
lvl1: 	System "coco" has no games! Ignoring it.
lvl1: 	  Unknown platform for system "dragon32" (platform "dragon32" from list "dragon32")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/dragon32/gamelist.xml"...
lvl1: 	System "dragon32" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/dreamcast/gamelist.xml"...
lvl1: 	System "dreamcast" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/fba/gamelist.xml"...
lvl1: 	System "fba" has no games! Ignoring it.
lvl1: 	  Unknown platform for system "fds" (platform "fds" from list "fds")
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/fds/gamelist.xml"...
lvl1: 	System "fds" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gamegear/gamelist.xml"...
lvl1: 	System "gamegear" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gb/gamelist.xml"...
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gba/gamelist.xml"...
lvl1: 	System "gba" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/gbc/gamelist.xml"...
lvl1: 	System "gbc" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/intellivision/gamelist.xml"...
lvl1: 	System "intellivision" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mame-advmame/gamelist.xml"...
lvl1: 	System "mame-advmame" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mame-libretro/gamelist.xml"...
lvl1: 	System "mame-libretro" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mame-mame4all/gamelist.xml"...
lvl1: 	System "mame-mame4all" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/mastersystem/gamelist.xml"...
lvl1: 	System "mastersystem" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/megadrive/gamelist.xml"...
lvl1: 	System "megadrive" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/msx/gamelist.xml"...
lvl1: 	System "msx" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/n64/gamelist.xml"...
lvl1: 	System "n64" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/neogeo/gamelist.xml"...
lvl1: 	System "neogeo" has no games! Ignoring it.
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/nes/gamelist.xml"...
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/ngp/gamelist.xml"...
[... 82 lines cut ...]
	</inputConfig>
</inputList>
```
```
$ cat /home/pi/.emulationstation/es_input.cfg
<?xml version="1.0"?>
<inputList>
  <inputAction type="onfinish">
    <command>/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh</command>
  </inputAction>
  <inputConfig type="joystick" deviceName="Logitech Gamepad F310">
    <input name="pagedown" type="button" id="5" value="1"/>
    <input name="start" type="button" id="7" value="1"/>
    <input name="pageup" type="button" id="4" value="1"/>
    <input name="up" type="hat" id="0" value="1"/>
    <input name="a" type="button" id="0" value="1"/>
    <input name="b" type="button" id="1" value="1"/>
    <input name="down" type="hat" id="0" value="4"/>
    <input name="right" type="hat" id="0" value="2"/>
    <input name="select" type="button" id="6" value="1"/>
    <input name="left" type="hat" id="0" value="8"/>
  </inputConfig>
</inputList>
```
Still, upon starting a game "Controller XYS not configured" is displayed in yellow for a second and the controls don't work.

joolswills, 2015-12-27:
Thanks - so the temporaryinput file is generated - not sure how you ended up with a blank one before. Please can you again run the input configuration a mention in my first reply, and send me the contents of any files in /opt/retropie/configs/all/retroarch-joypads

nschloe, 2015-12-27:
```
$ /opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
Input type is 'joystick'.
Configuring 'emulationstation'
Configuring 'mupen64plus'
Configuring 'reicast'
Configuring 'retroarch'
```
```
$ ls -l /opt/retropie/configs/all/retroarch-joypads/
total 8
-rw-r--r-- 1 pi pi 843 Dec 27 16:38 LogitechGamepadF310.cfg
-rw-r--r-- 1 pi pi 843 Dec 27 16:29 LogitechGamepadF310.cfg.bak
```
```
$ cat /opt/retropie/configs/all/retroarch-joypads/LogitechGamepadF310.cfg
input_device = "Logitech Gamepad F310"
input_driver = "udev"
input_r_y_plus_axis = "+4"
input_l3_btn = "9"
input_r_x_minus_axis = "-3"
input_r_btn = "5"
input_save_state_btn = "5"
input_l2_axis = "-2"
input_start_btn = "7"
input_exit_emulator_btn = "7"
input_l_y_minus_axis = "-1"
input_l_btn = "4"
input_load_state_btn = "4"
input_up_btn = "h0up"
input_r_y_minus_axis = "-4"
input_a_btn = "0"
input_b_btn = "1"
input_reset_btn = "1"
input_down_btn = "h0down"
input_l_x_plus_axis = "+0"
input_l_y_plus_axis = "+1"
input_r3_btn = "10"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_x_btn = "2"
input_menu_toggle_btn = "2"
input_select_btn = "6"
input_enable_hotkey_btn = "6"
input_l_x_minus_axis = "-0"
input_y_btn = "3"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_r_x_plus_axis = "+3"
```

joolswills, 2015-12-27:
Thanks. So you have the controller configuration correctly configured for retroarch. It should pick it up.

I will need to see the output from retroarch starting up. Please put a nes rom on the system and from a terminal via ssh run
```
/opt/retropie/emulators/retroarch/bin/retroarch --verbose -L /opt/retropie/libretrocores/lr-fceumm/fceumm_libretro.so --config /opt/retropie/configs/nes/retroarch.cfg /PATH/TO/NESROM
```

nschloe, 2015-12-27:
```
/opt/retropie/emulators/retroarch/bin/retroarch --verbose -L /opt/retropie/libretrocores/lr-fceumm/fceumm_libretro.so --config /opt/retropie/configs/nes/retroarch.cfg ~/RetroPie/roms/nes/Ninja\ Gaiden\ \(USA\).nes > out.log 2>&1
```
http://chunk.io/f/b7ac2108b8e54215963c24fe7ae51862

joolswills, 2015-12-27:
hmm, it's looking for another name it seems via retroarch. Does it make any difference if you switch the joypad from "xinput" to "direct input" ?

does it help if you switch the input driver to "udev" - to do this use the "Edit RetroPie/RetroArch configurations" from retropie-setup - choose "Change common RetroArch options" then /opt/retropie/config/all/retroarch.cfg and change input_joypad_driver to "udev"

joolswills, 2015-12-27:
Also another thing to try is with xboxdrv - Update retropie-setup script - go to "Xbox / Xbox 360 gamepad driver" from retropie-setup and choose "Enable xboxdrv" - then reboot and reconfigure in emulationstation again (as it should show up as a different device name)

nschloe, 2015-12-27:
> Does it make any difference if you switch the joypad from "xinput" to "direct input" ?

How do I do this?

> does it help if you switch the input driver to "udev"

The yellow font notification now says "Controler XYZ configured", and for some games it seems to work (NES Ninja Gaiden), some others work inconsistently (SNES Super Mario World, D-pad doesn't work), and in some others, the controls don't work at all (SNES Super Mario Kart).

nschloe, 2015-12-27:
> Also another thing to try is with xboxdrv -

That appears to work, but always recognizes two controllers. No problem in one-player games, but in two-player games, this becomes an issue on the "Player 2, select character" screens.

joolswills, 2015-12-27:
sounds like the udev driver might be the fix - but I don't know why it would only work for some - that's odd. 

Regarding the controller mode - Isn't there a button on the buttom of your controller to switch the mode ? Should be in the manual ?

chthonik, 2015-12-27:
Using the xboxdrv with the logitech 310 worked for me on snes anyways.

chthonik, 2015-12-27:
there is a 2nd mode for the 310 controller

nschloe, 2015-12-27:
> Regarding the controller mode - Isn't there a button on the buttom of your controller to switch the mode? Should be in the manual?

Right, there is. Never really understood what it's about. :)

> sounds like the udev driver might be the fix - but I don't know why it would only work for some - that's odd.

I'll try again.

nschloe, 2015-12-28:
I found that when setting the driver to _udev_ and physically reconnecting the controllers after a game has started, it all works as expected.

joolswills, 2015-12-28:
does it help to press the right shoulder button on this controller ? see https://www.reddit.com/r/RetroPie/comments/3yh7o3/pi_zero_latest_retropie_logitech_f310/ and the answers

nschloe, 2015-12-28:
Nope.

joolswills, 2015-12-28:
And switching the input mode of the joypad ? I'd be interested in results using sdl2 input driver as it seems others have been ok with that without having to switch to udev - perhaps related to the joypad mode?

nschloe, 2015-12-28:
With _sdl2_, it's all working fine in ES, but as soon as a game starts, I'm getting
```
Logitech F310 Gamepad (XInput) (1133/49693) not configured
```
and the controls don't work.

All of this is with the Logitech XD-swich on _X_. For _D_, the device isn't recognized at all, not even in ES.

alexbain, 2015-12-28:
Just wanted to chime in that I'm experiencing the same problem. I've tried disconnecting and reconnecting the gamepad after the game has started, but that hasn't worked for me either.

nschloe, 2015-12-28:
@alexbain does it say
```
Logitech F310 Gamepad (XInput) (1133/49693) not configured
```
for you at start up? Have you tried setting the input driver to udev? (The default SDL2 doesn't work for me either.)

alexbain, 2015-12-28:
Yeah, it says exactly that. Tried setting it to `udev`. Copied your config, too.

alexbain, 2015-12-28:
@nschloe - got it working. Try doing this:

edit `/opt/retropie/configs/all/retroarch.cfg`
go to line `338` change to `udev`

restart `emulationstation`

nschloe, 2015-12-28:
@alexbain The line you're referring to is already set to `udev` for me. That's how I got it to work with replugging after game start. Does it work from you without replugging?

alexbain, 2015-12-28:
Ah, interesting. No - I don't need to unplug/replug. Works fine on load.

nschloe, 2015-12-28:
What version of retropie are you running exactly?

alexbain, 2015-12-28:
Running off of master, latest commit is `b3c2ded4580dfd2c3541800a0d8badb60e3eb4c0`

alexbain, 2015-12-28:
I found that the f310 config file didn't match what was generated when I ran through the config. While not exactly related to this, here is the MR I created for the working f310 config I now have:

https://github.com/libretro/retroarch-joypad-autoconfig/pull/134

HypotomooseMan, 2015-12-30:
I had the same issue on Pi 2B with logitech F310 controller - changed the input driver from SDL2 to UDEV on line 338 and controller is working in NES & SNES Emulators.  I'm running retropie Weezy Standard 3.3.1.  Thanks for the help, let me know if i can be of any help.

gizmo98, 2015-12-30:
Seems to be sdl2 joypad driver has more cons than pros. If this assumption is true we should switch back to udev as default.

joolswills, 2015-12-30:
I have switched it.

Marclh97, 2016-01-02:
I've been killing myself trying to figure out exactly this. 
I have a rumblepad 2, raspberry pi 2 running jesse, and the latest retropie. 

I can't for the life of me get it working in a ROM. It works in emulationstation just fine. But I also can't get any file to appear in /opt/retroarch/configs/all/retroarch-joypads/

I've switched the driver to udev in half a dozen ways and back to sdl2 and rebooted and retried, and back to udev. Right now everything is set to udev. If I unplug and re-plug in game I get the "controller logitech logitech rumble pad 2 USB (something else) not configured" message anew.

I tried installing the xbox drivers for an experiment based on some comments, but now I just have 3 messages regarding controllers not being configured. Any help would be appreciated. 

Thanks

biscuits99, 2016-01-02:
Just to confirm the messages in this thread, I have had several comments on my YouTube videos about sdl2 causing the issues described as above, udev fixes it.
Although without the retroarch command line option available to re-configure the joypad config file, I'm not sure how to best go about correcting the invalid controller file that is generated via ES (for those controllers that like sdl2 like 8bitdo).

gizmo98, 2016-01-02:
Hi floob. I added a workaround for 8Bitdo controllers and udev. Update RetroPie-Setup and reinstall emulationstation if you would be so kind to test this functionality. Input_joypad_driver has to be udev.

biscuits99, 2016-01-02:
I'll give this a go on the Wheezy version now.

biscuits99, 2016-01-02:
- Booted Clean Wheezy 3.3.1
- Quit ES at configure prompt to go to command line
- Expanded FS, and rebooted, went back to CLI
- Update RetroPie Setup script via menu in ./retropie_setup.sh
- Then installed ES (binary) via retropie setup script (option 306)

Worked through the bluetooth setup steps 1-4 as written here:
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller

The FC30 Pro I used worked fine first time.

Here is the /opt/retropie/configs/all/retroarch-joypads/8BitdoFC30Pro.cfg
http://pastebin.com/raw/xBNNuqcw

Although I didnt change the input driver  
   # Input driver. Depending on video driver, it might force a different input driver.
   # input_driver = sdl  
  
   # Input device driver. (Valid: linuxraw, sdl, dinput)
   input_joypad_driver = sdl2  

Then I added the wired iBuffallo and they both work fine together.

Although when starting the rom, they both seem to be detected twice.
See video:
https://youtu.be/ImpQq9oRlfI

Should I try this again but change input_joypad_driver = sdl2 to udev before configuring the joypads?

I have attached ~/.emulationstation at the state it was in after adding the FC30 but before the iBuffallo.
[emulationstation.zip](https://github.com/RetroPie/RetroPie-Setup/files/76371/emulationstation.zip)

gizmo98, 2016-01-02:
Please change input driver to udev:
# Input device driver. (Valid: linuxraw, sdl, dinput)
input_joypad_driver = udev

Input config script decides what to do according to this setting.

biscuits99, 2016-01-03:
- Booted Clean Wheezy 3.3.1
- USB connections are, keyboard, bluetooth dongle
- Quit ES at configure prompt to go to command line
- Expanded FS, and rebooted, went back to CLI
- Update RetroPie Setup script via menu in ./retropie_setup.sh
- Then installed ES (binary) via retropie setup script (option 306)

- Updated /opt/retropie/configs/all/retroarch.cfg to
input_joypad_driver = udev

Worked through the bluetooth setup steps 1-4 as written here:
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller

The FC30 Pro I used worked fine first time.
Although the hotkeys dont work - this is explained by the mismatch button code entries in the following file.

Here is the /opt/retropie/configs/all/retroarch-joypads/8BitdoFC30Pro.cfg
http://pastebin.com/raw/VpKujETU

Then I added the wired iBuffallo and they both work fine together.

I still get a brief error when starting roms of
`Didnt read the whole file: sys/devices/system/cpu/present`
`Didnt read the whole file: sys/devices/system/cpu/possible`

I have attached ~/.emulationstation at the state it was in after adding the FC30 but before the iBuffallo.
[emulationstation.zip](https://github.com/RetroPie/RetroPie-Setup/files/76618/emulationstation.zip)

I have attached ~/.emulationstation at the state it was in after adding the FC30 and the iBuffallo.
[emulationstation-after.zip](https://github.com/RetroPie/RetroPie-Setup/files/76619/emulationstation-after.zip)

gizmo98, 2016-01-03:
Looks like every button >= 17 should not be touched. What a hack ;-)

gizmo98, 2016-01-03:
@biscuits99 / Floob
You could add a if clause around line 249 of retroarch.sh inputconfig script.
https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/emulationstation/configscripts/retroarch.sh#L249

This script can be found under /opt/retroPie/supplementery/emulationstation/configscripts/retroarch.sh
```
if [[ "$input_id" -lt "17" ]]; then
    input_id=$(($input_id+11))
fi
```

biscuits99, 2016-01-03:
- Booted Clean Wheezy 3.3.1
- USB connections are, keyboard, bluetooth dongle
- Quit ES at configure prompt to go to command line
- Expanded FS, and rebooted, exited ES, went back to CLI
- Update RetroPie Setup script via menu in ./retropie_setup.sh
- Then installed ES (binary) via retropie setup script (option 306)

- Updated /opt/retropie/configs/all/retroarch.cfg to
input_joypad_driver = udev

- Updated /opt/retropie/supplementary/emulationstation/scripts/configscripts/retroarch.sh so that the new if statement is in there.

It now reads
http://pastebin.com/raw/yKuC6ERE

Worked through the bluetooth setup steps 1-4 as written here:
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller

The FC30 Pro I used worked fine first time.
Although 2 hotkeys dont work (reset and RGUI) - this is explained by the mismatch button code entries in the following file.

Here is the /opt/retropie/configs/all/retroarch-joypads/8BitdoFC30Pro.cfg
http://pastebin.com/raw/C42CDnAB

I still get a brief error when starting roms of
`Didnt read the whole file: sys/devices/system/cpu/present`
`Didnt read the whole file: sys/devices/system/cpu/possible`

I have attached ~/.emulationstation here
[emulationstation.zip](https://github.com/RetroPie/RetroPie-Setup/files/76681/emulationstation.zip)

joolswills, 2016-01-03:
you can ignore the retroarch warnings - unrelated to controller etc

gizmo98, 2016-01-03:
@biscuits99  / Floob
I know why your hotkeys are wrong ;-)
You can transfer these changes to your config file:
https://github.com/RetroPie/RetroPie-Setup/commit/7127618090dd7c36866a8c5b5d130d1603b608fd

joolswills, 2016-01-03:
I would prefer input changes to be done on a branch first to avoid problems - especially when testing etc

gizmo98, 2016-01-03:
@joolswills 
Sorry. I thought i had compared at least most btn numbers (floob posted a good and bad config file). All buttons i had compared had an offset of 11. I didn't thought there "could be" a "step" after btn number 16.

joolswills, 2016-01-03:
NP. I can't see the functionality difference with the commit referenced above btw ? What have I missed ?

joolswills, 2016-01-03:
aah I see now, because it loops through the input_id should not be changed as they are used for multiple things.

biscuits99, 2016-01-03:
All working well now. Thanks :+1: 

Here is the new controller file.
http://pastebin.com/raw/7fGSx8Dd

~/.emulationstation attached
[emulationstation.zip](https://github.com/RetroPie/RetroPie-Setup/files/76745/emulationstation.zip)

joolswills, 2016-01-03:
I will rebuild retroarch and look to push out a new image in the next few days. Ideally I'd like to just move to Jessie - have either of you been testing with Jessie at all - I've not had any jessie specific bug reports I think.

joolswills, 2016-01-03:
Thanks for the fixes @gizmo09

biscuits99, 2016-01-03:
I've been using Jessie with no issues, but not tested specifics. Happy to run through a test plan for future releases.

HerbFargus, 2016-01-03:
I'm downloading a fresh jessie image for testing now to give it another go,  but I didn't run into anything major my first go through

[14 more replies]

closed 2016-02-22 by joolswills