RetroPie/RetroPie-Setup #1156: Controller configuration xbox 360 wired

Javison, 2015-12-27:
Hi,

Iam having problems to configure this gamepad. Using 3.3.1 image new install on Raspberry 1
After use Xbox / Xbox 360 gamepad driver" option in the Setup / Configuration menu of the retropie-setup script, there is no such folder: "/opt/retropie/supplementary/xboxdrv"

The file /etc/rc.local contains:
"/opt/retropie/supplementary/xboxdrv/bin/xboxdrv" --daemon --detach --dbus disabled --detach-kernel-driver --id 0 --led 2 --deadzone 4000 --silent --trigger-as-button --next-controller --id 1 --led 3 --deadzone 4000 --silent --trigger-as-button
exit 0

Disable/re-eable, and reboot several times.

Thanks,

joolswills, 2015-12-27:
No idea why the xboxdrv hasnt installed as it should - maybe a network problem when it was installing or something. If the folder isn't there it should install the first time when launching the configuration.

from a terminal please do

```
cd
cd RetroPie-Setup
sudo ./retropie_packages.sh xboxdrv install_bin
```

Javison, 2015-12-27:
pi@retropie ~/RetroPie-Setup $ sudo ./retropie_packages.sh xboxdrv install_bin
^[[D
^[[D^[[C

^[[D

= = = = = = = = = = = = = = = = = = = = =
Installing binary archive for Xbox / Xbox 360 gamepad driver
= = = = = = = = = = = = = = = = = = = = =

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now

joolswills, 2015-12-27:
do you have an internet connection ?

Javison, 2015-12-27:
Yes. :)

joolswills, 2015-12-27:
Also - the 3.3.1 image comes with xboxdrv already installed - so I don't know why it wouldn't be there - are you sure the folder is missing?

what is the output of 

```
ls -l /opt/retropie/supplementary/xboxdrv
```

you definitely started with a fresh image ? which image specifically ? Wheezy or Jessie / Berry boot etc

joolswills, 2015-12-27:
Aha. I see the problem.

Javison, 2015-12-27:
pi@retropie ~ $ pwd
/home/pi
pi@retropie ~ $ ls -l /opt/retropie/supplementary/xboxdrv
ls: cannot access /opt/retropie/supplementary/xboxdrv: No such file or directory
pi@retropie ~ $

joolswills, 2015-12-27:
Looks like xboxdrv package is missing for rpi1 - must have been missed when building everything for release -  ill fix that now.

Javison, 2015-12-27:
Thanks a lot. Looking forward!

joolswills, 2015-12-27:
ok please try the install_bin now (or just disable/re-enable from retropie-setup)

Javison, 2015-12-27:
THnaks, The folder created succesfully,

pi@retropie ~/RetroPie-Setup $ sudo ./retropie_packages.sh xboxdrv install_bin

= = = = = = = = = = = = = = = = = = = = =
Installing binary archive for Xbox / Xbox 360 gamepad driver
= = = = = = = = = = = = = = = = = = = = =

xboxdrv/
xboxdrv/share/
xboxdrv/share/man/
xboxdrv/share/man/man1/
xboxdrv/share/man/man1/xboxdrv.1
xboxdrv/bin/
xboxdrv/bin/xboxdrv
xboxdrv/bin/xboxdrvctl

closed 2015-12-28 by Javison