RetroPie/RetroPie-Setup #1157: lr-ppsspp Not Reading Config File

craigerskine, 2015-12-28:
I'm trying to override some settings from the default retroarch config for PSX and PSP... it works for PSX but not PSP.

I think the file from /opt/retropie/configs/psp/retroarch.cfg is not being read. I use the same lines in both PSX and PSP configs:

    video_shader_enable = false
    rewind_enable = false

Again, it works in PSX but not PSP. And yes, when I launch a game it is indeed using lr-ppsspp.

joolswills, 2015-12-29:
Please use the RetroPie forum for help. https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2015-12-29 by joolswills

craigerskine, 2016-01-04:
I don't need help, I already know how it is supposed to work, I'm reporting a bug.

joolswills, 2016-01-04:
I believe it's like a configuration issue your end hence suggesting you use the forum first. if reporting a bug, I will require the complete configurations etc (can use pastebin.com or similar for large files).

joolswills, 2016-01-04:
configs/all/retroarch.cfg
configs/psp/retroarch.cfg
configs/psp/emulators.cfg please.