RetroPie/RetroPie-Setup #1159: Configure Retroarch Controller Missing

marvinej, 2015-12-29:
The Configure Retroarch controller option is missing from my retropie-setup thus i am unable to use my snes controller by following the instructions in the wiki

running the latest version

gizmo98, 2015-12-30:
Which wiki page do you refer to? Retroarch guys removed retroarch-joyconfig a long time ago. There is no retroarch-joyconfig anymore. Controllers should be setup through emulationstation.

closed 2015-12-30 by gizmo98

marvinej, 2015-12-30:
The video you link to for configuring usb controllers says to use this
option. If this is no longer required then why doesnt the NES work with my
usb snes controller out of the box? It wont work with the xbox controller
either when i install xboxdrv. The main interface works but the emulators
dont.

joolswills, 2015-12-30:
please reference the wiki page in question. thanks.

HerbFargus, 2015-12-30:
The wiki is created by users, and most videos are outdated as the retroarch-joyconfig was a somewhat recent removal. I will make an updated video at some point (hopefully soon) and update the wiki