RetroPie/RetroPie-Setup #1160: Splash video crashing after update to 3.3

markyh444, 2015-12-30:
I did an upgrade to version 3.3 from 3.2.1 from the retropie_install.sh in order to resolve some n64 issues, but now my splash start-up video crashes halfway through and the pi won't load into emulationstation. If I swap for a default, or other static splash, it loads fine. I've tried removing the videos, disabling custom splashes and restarting and setting up again, but with the same results. Any help would be great as I had it booting lovely before the update.

joolswills, 2015-12-30:
What is retropie_install.sh ?

markyh444, 2015-12-30:
Sorry, was typing this when my son was crying at me. I meant retropie_setup.sh

joolswills, 2015-12-30:
Not sure what the problem could be - something with your video file ?

Please post the output of `md5sum /etc/init.d/asplashscreen`
and
the contents of /etc/splashscreen.list (as well as a link to the file referenced)

markyh444, 2015-12-30:
I'll do that later (not at home at the minute) but it was all working fine with the last release so I don't think it's the video file. Unless it suddenly doesn't support mp4 files now.

joolswills, 2015-12-30:
well, a firmware or omxplayer update could have affected it

markyh444, 2015-12-30:
No probs, appreciate the guidance. I'll get the logs up later today and take a look to see if there's a mention of an issue with omxplayer.

joolswills, 2015-12-30:
it could be due to a parameter change also in the way we call omxplayer specifying a layer, but I can advise once I have the files. thanks.

markyh444, 2015-12-30:
Hi, I've managed to run through this now.

Here's the contents of the splashscreen.list - /home/pi/RetroPie/splashscreens/video/nespie.mp4

Here's a link to the video - https://www.dropbox.com/sh/t1rvr7ya1t0fgqx/AAAdINXOg0icCPXUz0Raig_xa?dl=0 

And here is the output of md5sum /etc/init.d/asplashscreen - 7f6e46c67a78c8ee260a51ada28e8e90  /etc/init.d/asplashscreen

I've had a little root through that file as well just to make sure that it tallied in with what the older one that worked looked like and it does have the line omxplayer -b --layer 10000 "$line" which should put it at the correct layer as far as I understand. 

I mean, the video does start and it runs for about 10 seconds, then glitches to a black screen. It does come back on occasion, but frozen where it left off, and it may flick back and forth to the same bit every minute or so. Quite odd...

joolswills, 2015-12-30:
Raspbian Wheezy or Jessie ?

markyh444, 2015-12-30:
It's Wheezy.

joolswills, 2015-12-30:
To rule out an issue with your set up (such as from installing additional software etc / config changes), please can you download a fresh 3.3.1 image and try it with the video - if you still have the problem I will test it myself on a fresh image. thanks.

markyh444, 2016-01-01:
Hi, just to let you know that a fresh install seems to have cured it. Not sure what I did to break it in the first case, but I'm not complaining if it works now. Thanks for your help and keep up the good work.

closed 2016-01-01 by markyh444