RetroPie/RetroPie-Setup #1162: Wonderswan Save Issue

Omnija, 2015-12-31:
Wonderswan and Color save games in the bios folder.

These lines should be added to each individual emulators retroarch.cfg to make sure they save in the game folder.
system_directory = /home/pi/RetroPie/roms/wonderswancolor
system_directory = /home/pi/RetroPie/roms/wonderswan

Neogeo Pocket and Color save in the home folder still unsure why or how to redirect them.

gizmo98, 2015-12-31:
Sounds like a core problem because system_directory is for bios files https://github.com/libretro/RetroArch/blob/master/retroarch.cfg#L74

Omnija, 2015-12-31:
I believe it's the "mednafen" emulators that tend save where they please.

gizmo98, 2015-12-31:
What happens if you set `savefile_directory = /home/pi/RetroPie/roms/wonderswan`?

Omnija, 2015-12-31:
nothing, i've already tried multiple times this was the only method that worked. I still yet to find a method to get neogeo pocket working.

gizmo98, 2015-12-31:
Ok. I had seen some lines of code suggesting this:
https://github.com/libretro/beetle-wswan-libretro/blob/master/libretro.cpp#L542

On the other hand i do not understand why system_directory is base_directory:
https://github.com/libretro/beetle-wswan-libretro/blob/master/libretro.cpp#L524

Omnija, 2015-12-31:
As for the Neogeo Pocket emulators, i don't think they are even declared a location to save. Which is probably why they can't be directed in any way.

HerbFargus, 2016-01-03:
we had the same issue with gpsp, save states only work in the directory it's installed in

joolswills, 2016-01-03:
@HerbFargus that wasn't a libretro emulator issue though afair

HerbFargus, 2016-01-03:
Ah, yeah i suppose you're right.

gizmo98, 2016-01-10:
@joolswills @HerbFargus 
In my opinion we could set system_directory for both systems if wonderswan needs no bios file?!

Omnija, 2016-01-10:
That's what i did to correct wonderswan systems but didn't work for Neogeo Pocket/Colour  (saves home folder of pi)

HerbFargus, 2016-04-21:
@Omnija 

For NeoGeo Pocket try these fixes for the save dir:

https://github.com/HerbFargus/beetle-ngp-libretro/commit/590b6b496b65d31458351840e221e3cf4b16911d

HerbFargus, 2016-04-21:
@Omnija try out the fix on my branch and tell me if it fixes the issue:

https://github.com/HerbFargus/beetle-ngp-libretro/tree/save-fix

HerbFargus, 2016-04-22:
the `savefile_directory` code still needs to be sorted upstream but now in game saves will at least default to the rom directory now. same for NGP

closed 2016-04-22 by HerbFargus