RetroPie/RetroPie-Setup #1166: $dispmanx_conf is ignored on some systems

mruwek, 2016-01-04:
I've installed RetroPie via binary installation on [OSMC](https://osmc.tv) with help of [mcobit/retrosmc](https://github.com/mcobit/retrosmc) script. Everything works well so far except for emulators/ports that need dispmanx despite having it properly configured in `$configdir/all/dispmanx.cfg`.

My example is OpenTyrian
Running it from EmulationStation always ends with such entry in `/tmp/runcommand.log`:
> `failed to initialize any supported video`

1. I've read in [discussion](https://github.com/RetroPie/RetroPie-Setup/issues/1050#issuecomment-148933668) under #1050 that this is caused by unset dispmanx flag.
2. I double-checked file permissions of `$configdir/all/dispmanx.cfg` and its contents. File permissions were fine, and `dispmanx.cfg` had line:
> opentyrian="1"

3. I've also added to `runcommand.sh` tiny debug-purpose line `echo $save_emu > /tmp/save_emu` right after calling of `get_save_vars`. Resulting file after starting the game from ES was blank.
4. Then I've tried also to comment out [line 91](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/runcommand/runcommand.sh#L91) and after that OpenTyrian works flawlessly and `/tmp/save_emu` has one line with 'opentyrian` in it.

It seems then that the problem lies in this particular substitution:

```
save_emu=${save_emu//[^a-Z0-9_\-]/}
```

Right now I don't have any idea why this substitution don't work. Maybe it's related with some bash-4.3 changes?

#### Sidenote:
Before installation of OpenTyrian I had to:
- install **mercurial** via `apt-get install mercurial` - `retropie_setup.sh` didn't succeed with that for some reason.
- install **unzip** via `apt-get install unzip` - default version of unzip on OSMC doesn't have option `-j` used by `retropie_setup.sh` during extraction of *tyrian21.zip*.

joolswills, 2016-01-04:
Ignore the reference to dispmanx flag in that discussion. Opentyrian works with and without the dispmanx sdl backend. It should be faster without.

I suspect your problems might be related to the way retrosmc installs everything. Have you asked on the OSMC thread about this ? I cannot really advise on 3rd party implementations (I'm not sure if retrosmc is still using a chroot method of installation or not)

save_emu must be working as it affects the dispmanx setting and is making a difference. I think it's an issue with sdl - Your error actually sounds like either you dont have our sdl installed, or something else.

I recommend you try running it manually from terminal 

```
standard sdl framebuffer output:
/opt/retropie/ports/opentyrian/bin/opentyrian --data ~/RetroPie/roms/ports/opentyrian/data/
force dispmanx sdl1 backend:
SDL1_VIDEODRIVER="dispmanx"  /opt/retropie/ports/opentyrian/bin/opentyrian --data ~/RetroPie/roms/ports/opentyrian/data/
```

Regarding other problems - Are you using busybox or something ? which unzip was being used before ? I can only think you are using some shell with unzip built in by default ? I will add unzip as a dependency though. Can't help with the installation problem of mercurial without logs.

I would add that OSMC support is still to be considered WIP, and I have only tested myself installing retropie modules directly without any additional helper scripts like retrosmc.

mruwek, 2016-01-04:
> I cannot really advise on 3rd party implementations

Sure thing. Consider this issue report as a try to help with providing better support for OSMC :) I know very well that for now I'm on my own.

>  I'm not sure if retrosmc is still using a chroot method of installation or not

Latest version isn't using chroot. (alpha 0.007)

> save_emu must be working (...)

It's working (i.e. non-empty) **only when** I delete or comment out [line](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/runcommand/runcommand.sh#L91) mentioned above. Look at my debug approach - blank `$save_emu` with this line enabled and `opentyrian` with this line commented out. Furthermore - game starts right after commenting this line (`$save_emu` is then successfully set to `opentyrian`). Even more furthermore - with this line enabled I had problems in emulator/rom config menu (accessed with button0 right after start) - trying to set default video/framebuffer also ended with errors. And again no problems after commenting out. Seriously - it's something with bash-4.3 or the way that bash is called from RetroPie on OSMC.

I don't understand why it fails, because this line is in my opinion completely sane, but it fails nonetheless when called from RetroPie. I can post a screencast if you want me to prove it :)

> Are you using busybox or something ? which unzip was being used before ?

Yep, OSMC uses busybox and Dash AFAIK. Which can be an answer to problem above. As we read in https://wiki.ubuntu.com/DashAsBinSh:
> **${parm/?/pat[/str]}**
>
> The '${parm/?/pat[/str]}' warning from the checkbashisms script.
>
> This is a parameter expansion ${parameter/pattern/string} supported in bash but not in dash. To replace these, you could make use of echo, sed, grep, and/or awk among other tools. For example.

I'll try to implement this line with sed or awk and see what happens then.

mruwek, 2016-01-04:
>  I will add unzip as a dependency 

Great :)

> Can't help with the installation problem of mercurial without logs.

I'll try to collect them and post them here.

joolswills, 2016-01-04:
RetroPie is all bash based and expects bash to be installed. When I was testing on OSMC it wasn't using buybox so has that changed or I was using another version. Is bash installed or does "bash" launch busybox ?

joolswills, 2016-01-04:
I wont be interested in removing bashisms - we are specifically using bash features btw.

mruwek, 2016-01-04:
Bash (4.3) is definitely installed and as I wrote previously - testing `get_save_vars` (copying it to another file,sh and hardcoding `$emulator`) on the same machine (via ssh) worked well.

joolswills, 2016-01-04:
Somehow runcommand is being launched without bash then ? What doe "echo $0" show if you stick it at the top of the script ?

mruwek, 2016-01-04:
Looks like problem exists only when `runcommand.sh` is called from within EmulationStation.

joolswills, 2016-01-04:
If you can work out why it's happening that would help. I can't really advise without being able to reproduce the problem myself - which I can try when I have some free time to test again on OSMC. Also you might want to launch ES manually from terminal via /opt/retropie/supplementary/emulationsstation/emulationstation to rule out anything related to the way ES is launched from within Kodi or via another script (I know retrosmc has it's own launch scripts etc)

mruwek, 2016-01-04:
> When I was testing on OSMC it wasn't using buybox so has that changed or I was using another version.

I'm quite sure dash is the default system shell, but for interactive login shells (ssh) it could be bash (that would explain why `get_save_vars` works properly via ssh). I'm also really sure now that unzip shipped with OSMC is a busybox implementation.

> If you can work out why it's happening that would help

As soon as I'll be back home I'll try to test `echo $0`.

joolswills, 2016-01-04:
BTW just as some additional info -  it does work without dispmanx - just make sure you have a framebuffer mode configured for the output res. eg 640x400 (default when using x2 scalar - scalar can be switched off for possibly better performance). It works here, so perhaps OSMC doesn't have as many default framebuffer modes configured too.

joolswills, 2016-01-04:
Also you can add

```
echo $BASH_VERSION
```

I would like to know if exiting kodi to a terminal and manually launching ES works as mentioned above (not via retrosmc /usr/bin/emulationstation script).

mruwek, 2016-01-04:
Ok. Some good news and bad news.

1. Starting ES from ssh via retrosmc script: `./RetroPie/scripts/retopie.sh` works.
2. Starting via retrosmc Kodi launcher addon doesn't.

It's strange as this launcher does nothing more than:

```python
output=os.popen("/bin/bash/home/osmc/RetroPie/scripts/retropie.sh").read()
```

So this is almost surely not a RetroPie bug, nor bash version problem, nor retrosmc bug. Probably problem with some env var passed by Kodi addons framework or and messing up some bashisms in result.

joolswills, 2016-01-04:
http://www.saltycrane.com/blog/2011/04/how-use-bash-shell-python-subprocess-instead-binsh/

has the solution possibly.

mruwek, 2016-01-05:
> has the solution possibly.

I've tried that before with no result

mruwek, 2016-01-05:
Later today I'll post one small gist (test script containing `get_save_vars` function outputting `$save_emu` to tmpfile) that can be set in place of `/home/osmc/RetroPie/script/retropie.sh`) that can help to easily reproduce this bug without installation of OpenTyrian. I have also nearly completely excluded theory that `runcommand.sh` is run by other interpreter than Bash. I have two reasons for this:

1. There is another substring substitution in `the get_save_vars` at [line 90](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/runcommand/runcommand.sh#L90): `save_emu=${emulator//\//_}`. If whole script would be run by Dash, this line would also not work.
2. I tried to run `get_save_vars` function explicitly with Dash and it failed then with `Bad substitution` error. No such error when trying to run OpenTyrian from ES.

I'll also post an issue report at retrosmc tracker so @mcobit can share his thoughts.

mruwek, 2016-01-05:
Ok, found it. It's related with collate shell env-variable, but there is one simple change that makes the whole line totally immune to the environment it is run in. I'll provide PR in a moment.

joolswills, 2016-01-05:
aah so we can do A-Za-z instead and it will resolve it right ?

joolswills, 2016-01-05:
nice one btw :-)

mruwek, 2016-01-05:
> aah so we can do A-Za-z instead and it will resolve it right ?

Right :)

PR #1171 (2016-01-05): Use safer range expression in get_save_vars
a-Z depends on LC_COLLATE or LC_ALL and can produce weird bugs if user
have set there something other than what we're using by default.

a-zA-Z should work just everywhere.

Fixes #1166 and mcobit/retrosmc#14
commit ce186420 (2016-01-05): Use safer range expression in get_save_vars

a-Z depends on LC_COLLATE or LC_ALL and can produce weird bugs if user
have set there something other than what we're using by default.

a-zA-Z should work just everywhere.

Fixes #1166 and mcobit/retrosmc#14
closed 2016-01-05 by joolswills