RetroPie/RetroPie-Setup #1170: Xboxdrv Updates (Xbox One Controller Support and more)

zerojay, 2016-01-05:
I thought I would bring to your attention my fork of xboxdrv which basically consists of the latest xboxdrv as well as some of the current pending xboxdrv pull requests, in case you might be interested: https://github.com/zerojay/xboxdrv.git

#137 - Fix zero-length rumble packets causing infinite rumble on some controllers.
#169 - Add support for MadCatz Xbox 360 Marvel vs Capcom Tournament Edition Fightstick.
#173 - Fix minor typo.
#174 - Fix USB read error causing controller to be unusable until xboxdrv is restarted manually. (Happens somewhat frequently to me, personally.)
#175 - Add support for Xbox One controllers (rumble and LED lights not functional yet).

joolswills, 2016-01-09:
Thanks. We could switch temporarily perhaps, but upstream was doing some development near the end of last year so hopefully work will continue on the main repository.. Please submit a pull request to repoint to your repository and I will merge.

zerojay, 2016-01-10:
Done: https://github.com/RetroPie/RetroPie-Setup/pull/1180

If it needs to be updated elsewhere than xboxdrv.sh, I am not aware of it.

zerojay, 2016-01-11:
With the above pull request integrated, this can be closed.

closed 2016-01-11 by zerojay

granitrocky, 2025-04-30:
Is there a working fork of @crcerror 's xboxdrv? It looks like it's private now