RetroPie/RetroPie-Setup #1173: Bluetooth Controller Buttons Not Registering!

sentencebender, 2016-01-06:
Hey there. 

I got my bluetooth Nyko Playpad controller to pair with the Raspberry Pi with no problems, but when the EmulationStation controller button mapping wizard opens up, several key buttons simply don't register - when I press them, or even hold them, nothing happens... any thoughts on what to do? I tried switching to Mouse/Keyboard/App/Joystick modes on the PlayPad, and some buttons show up in one mode but not in others, but there's no mode that will get me through a full normal set-up that I can use for gaming, and some buttons don't register in any mode.

Thanks!!

joolswills, 2016-01-09:
Does `jstest /dev/input/jsX` show anything for those buttons ?

do you have the latest firmware ?

joolswills, 2016-03-03:
closed due to no reply.

closed 2016-03-03 by joolswills

hanthecat, 2016-03-30:
I have the same controller. I see most buttons work under jstest except for what should be the select button. emulationstation sees that as a keyboard press.

Also, Im running fbsp to get my waveshare screen working. You can see that behind the frambuffer copy the console is still active. I get "They key you just pressed is not recognized by SDL"

I have a ps3 controller that works great though that is not as ideal for travel as the Nyko Playpad.

Can someone point me in the right direction to get this guy working?