RetroPie/RetroPie-Setup #1177: Wrong joypad configuracion file

makmonty, 2016-01-08:
Hi there,

I have an issue with my 8Bitdo Nes30 Pro pad. When I configure it via the "Configure Input" option in the menu of EmulationStation, it writes wrong information in the 8BitdoNES30Pro.cfg file. It writes the following:

    input_device = "8Bitdo NES30 Pro"
    input_driver = "udev"
    input_r_y_plus_axis = "+3"
    input_l3_btn = "13"
    input_r_x_minus_axis = "-2"
    input_r_btn = "7"
    input_save_state_btn = "7"
    input_l2_btn = "8"
    input_start_btn = "11"
    input_exit_emulator_btn = "11"
    input_l_y_minus_axis = "-1"
    input_l_btn = "6"
    input_load_state_btn = "6"
    input_up_btn = "h0up"
    input_r_y_minus_axis = "-3"
    input_a_btn = "0"
    input_b_btn = "1"
    input_reset_btn = "1"
    input_down_btn = "h0down"
    input_r2_btn = "9"
    input_l_x_plus_axis = "+0"
    input_l_y_plus_axis = "+1"
    input_r3_btn = "14"
    input_right_btn = "h0right"
    input_state_slot_increase_btn = "h0right"
    input_x_btn = "3"
    input_menu_toggle_btn = "3"
    input_select_btn = "10"
    input_enable_hotkey_btn = "10"
    input_l_x_minus_axis = "-0"
    input_y_btn = "4"
    input_left_btn = "h0left"
    input_state_slot_decrease_btn = "h0left"
    input_r_x_plus_axis = "+2"

but it should be:

    input_device = "8Bitdo NES30 Pro"
    input_driver = "udev"
    input_b_btn = "12"
    input_y_btn = "15"
    input_select_btn = "21"
    input_start_btn = "22"
    input_up_btn = "h0up"
    input_down_btn = "h0down"
    input_left_btn = "h0left"
    input_right_btn = "h0right"
    input_a_btn = "11"
    input_x_btn = "14"
    input_l_btn = "17"
    input_r_btn = "18"
    input_l2_btn = "19"
    input_r2_btn = "20"
    input_l3_btn = "24"
    input_r3_btn = "25"
    input_l_x_plus_axis = "+0"
    input_l_x_minus_axis = "-0"
    input_l_y_plus_axis = "+1"
    input_l_y_minus_axis = "-1"
    input_r_x_plus_axis = "+2"
    input_r_x_minus_axis = "-2"
    input_r_y_plus_axis = "+3"
    input_r_y_minus_axis = "-3"
    input_enable_hotkey_btn = "21"
    input_exit_emulator_btn = "22"
    input_menu_toggle_btn = "14"
    input_load_state_btn = "17"
    input_save_state_btn = "18"
    input_reset_btn = "12"
    input_state_slot_increase_btn = "h0right"
    input_state_slot_decrease_btn = "h0left"

I have the latest version of RetroPie (updated yesterday) and I can't find a workaround or anything, I just would like it to work without headaches, but instead I have to SSH every time to replace the file with the right one. Ok, I could create an script to do it for me, but it's not the solution I'm looking for.

Thanks for your help.

gizmo98, 2016-01-08:
Please do a source based installation of emulationstation.

makmonty, 2016-01-09:
Done. No changes.

joolswills, 2016-01-09:
did you reconfigure your controller after updating and reinstalling emulationstation? (you can re-install emulationstation from binary - it needs to be reinstalled to trigger the script update)

makmonty, 2016-01-09:
Yes, after the compilation and  installation I rebooted the system and then configured the pad again, but the file created is wrong

biscuits99, 2016-01-09:
Could the all/retroarch.cfg have sdl2 in it?

joolswills, 2016-01-09:
@makmonty what is the output of md5sum /opt/retropie/supplementary/emulationstation/scripts/configscripts/retroarch.sh

joolswills, 2016-01-09:
also please re-post the newly created configuration file for the joypad.

joolswills, 2016-01-09:
aah I don't think we have an entry for your model - we should add it. I think all the 8bitdo controllers are affected so maybe we should reverse our logic.

gizmo98, 2016-01-09:
Yes. 8Bitdo NES30 Pro is missing.

joolswills, 2016-01-09:
ok please update retropie-setup again, re-install ES from binary, and reconfigure controller. I have added your joypad now.

@gizmo98 maybe we should just match all 8Bitdo ?

biscuits99, 2016-01-09:
In case its relevant, there is a good collection of model names here:
https://github.com/libretro/retroarch-joypad-autoconfig/tree/master/udev

joolswills, 2016-01-09:
@biscuits99 - you have the arcade controller right ? Does that require the same hack with the incremented ids ?

gizmo98, 2016-01-09:
I compared some 8Bitdo controller configs from libretro's retroarch-joyconfig repo. Seems to be all of them use the same IC. I don't know if we should add this workaround for all 8Bitdo controllers. If one of them behaves different and does not need this fix we need a solution the other way around (whitelist/blacklist).

joolswills, 2016-01-09:
yep, im just thinking it might be easier the other way around - especially if new models come out that need the hack (then they would work by default). I suspect most will since they probably use a similar firmware for all the devices etc

biscuits99, 2016-01-09:
@joolswills From memory the sanwa bluetooth from 8bitdo needed the hack with bluetooth but not when used with USB. Havent tried with new script yet though.

joolswills, 2016-01-09:
thanks. hmm that will be fun to fix :)

biscuits99, 2016-01-09:
As an aside the 8bitdo devs, seem very good with suggestions of firmware changes. I'll point them to this thread.

makmonty, 2016-01-10:
It works perfect now, thank you guys. If there's something I can do to help you debugging the controller, just let me know.

johnjelinek, 2016-01-19:
I'm also having controller configuration issues with ES since upgrading (to source):

I'm using a Steam Controller that's acting like an xbox controller.
```
$ cat es_input.cfg
<?xml version="1.0"?>
<inputList>
  <inputConfig type="joystick" deviceName="Microsoft X-Box 360 pad">
    <input name="a" type="button" id="0" value="1"/>
    <input name="b" type="button" id="1" value="1"/>
    <input name="down" type="hat" id="0" value="4"/>
    <input name="left" type="hat" id="0" value="8"/>
    <input name="leftanalogdown" type="axis" id="1" value="1"/>
    <input name="leftanalogleft" type="axis" id="0" value="-1"/>
    <input name="leftanalogright" type="axis" id="0" value="1"/>
    <input name="leftanalogup" type="axis" id="1" value="-1"/>
    <input name="leftbottom" type="axis" id="2" value="-1"/>
    <input name="leftthumb" type="button" id="9" value="1"/>
    <input name="right" type="hat" id="0" value="2"/>
    <input name="rightanalogdown" type="axis" id="4" value="1"/>
    <input name="rightanalogleft" type="axis" id="3" value="-1"/>
    <input name="rightanalogright" type="axis" id="3" value="1"/>
    <input name="rightanalogup" type="axis" id="4" value="-1"/>
    <input name="rightbottom" type="axis" id="2" value="1"/>
    <input name="rightthumb" type="button" id="10" value="1"/>
    <input name="righttop" type="button" id="5" value="1"/>
    <input name="select" type="button" id="6" value="1"/>
    <input name="start" type="button" id="7" value="1"/>
    <input name="up" type="hat" id="0" value="1"/>
    <input name="x" type="button" id="2" value="1"/>
    <input name="y" type="button" id="3" value="1"/>
    <input name="pageup" type="axis" id="2" value="1"/>
  </inputConfig>
  <inputAction type="onfinish">
    <command>/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh</command>
  </inputAction>
</inputList>
```

```
$ cat es_temporaryinput.cfg
<?xml version="1.0"?>
<inputList>
	<inputAction type="onfinish">
		<command>/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh</command>
	</inputAction>
	<inputConfig type="joystick" deviceName="Microsoft X-Box 360 pad" deviceGUID="030000005e0400008e02000001000000">
		<input name="a" type="button" id="0" value="1" />
		<input name="b" type="button" id="1" value="1" />
		<input name="down" type="hat" id="0" value="4" />
		<input name="left" type="hat" id="0" value="8" />
		<input name="leftanalogdown" type="axis" id="1" value="1" />
		<input name="leftanalogleft" type="axis" id="0" value="-1" />
		<input name="leftanalogright" type="axis" id="0" value="1" />
		<input name="leftanalogup" type="axis" id="1" value="-1" />
		<input name="leftbottom" type="axis" id="2" value="1" />
		<input name="leftthumb" type="button" id="9" value="1" />
		<input name="lefttop" type="button" id="4" value="1" />
		<input name="right" type="hat" id="0" value="2" />
		<input name="rightanalogdown" type="axis" id="4" value="1" />
		<input name="rightanalogleft" type="axis" id="3" value="-1" />
		<input name="rightanalogright" type="axis" id="3" value="1" />
		<input name="rightanalogup" type="axis" id="4" value="-1" />
		<input name="rightthumb" type="button" id="10" value="1" />
		<input name="righttop" type="button" id="5" value="1" />
		<input name="select" type="button" id="6" value="1" />
		<input name="start" type="button" id="7" value="1" />
		<input name="up" type="hat" id="0" value="1" />
		<input name="x" type="button" id="2" value="1" />
		<input name="y" type="button" id="3" value="1" />
	</inputConfig>
</inputList>
```

johnjelinek, 2016-01-19:
note: the input config menu never seems to get my right bottom button correctly (it skips it after I do left bottom), so I've had to manually add that one so it looks like this:
```
$ cat es_input.cfg.bak
<?xml version="1.0"?>
<inputList>
	<inputConfig type="joystick" deviceName="Microsoft X-Box 360 pad" deviceGUID="030000005e0400008e02000001000000">
		<input name="a" type="button" id="0" value="1" />
		<input name="b" type="button" id="1" value="1" />
		<input name="down" type="hat" id="0" value="4" />
		<input name="left" type="hat" id="0" value="8" />
		<input name="leftanalogdown" type="axis" id="1" value="1" />
		<input name="leftanalogleft" type="axis" id="0" value="-1" />
		<input name="leftanalogright" type="axis" id="0" value="1" />
		<input name="leftanalogup" type="axis" id="1" value="-1" />
		<input name="leftbottom" type="axis" id="2" value="1" />
		<input name="leftthumb" type="button" id="9" value="1" />
		<input name="lefttop" type="button" id="4" value="1" />
		<input name="right" type="hat" id="0" value="2" />
		<input name="rightanalogdown" type="axis" id="4" value="1" />
		<input name="rightanalogleft" type="axis" id="3" value="-1" />
		<input name="rightanalogright" type="axis" id="3" value="1" />
		<input name="rightanalogup" type="axis" id="4" value="-1" />
		<input name="rightbottom" type="axis" id="5" value="1" />
		<input name="rightthumb" type="button" id="10" value="1" />
		<input name="righttop" type="button" id="5" value="1" />
		<input name="select" type="button" id="6" value="1" />
		<input name="start" type="button" id="7" value="1" />
		<input name="up" type="hat" id="0" value="1" />
		<input name="x" type="button" id="2" value="1" />
		<input name="y" type="button" id="3" value="1" />
	</inputConfig>
</inputList>
```

joolswills, 2016-01-19:
you can go "up" when you get to "ok" to correct the other inputs.

What problem are you having ?

johnjelinek, 2016-01-19:
in ES, I can navigate fine with the controller, but in any of the emulators, none of the buttons are being recognized although the yellow text says it recognizes an xbox controller on `port #0`.

joolswills, 2016-01-19:
please post contents (via pastebin.com or similar as large ) of /opt/retropie/configs/all/retroarch.cfg - and anything in /opt/retropie/configs/retroarch-joypads

joolswills, 2016-01-19:
also check this and solution in case it's the same problem - https://github.com/RetroPie/RetroPie-Setup/issues/1140

johnjelinek, 2016-01-19:
my input driver is already `udev`.

johnjelinek, 2016-01-19:
retroarch.cfg - http://pastebin.com/akpgy72H

I don't have a `/opt/retropie/configs/retroarch-joypads`, but if you mean, to pick out a config for an emulator where I'm having trouble:

```
pi@raspberry:/opt/retropie/configs/megadrive $ cat retroarch.cfg
# Settings made here will only override settings in the global retroarch.cfg if placed above the #include line

input_remapping_directory = /opt/retropie/configs/megadrive/

#include "/opt/retropie/configs/all/retroarch.cfg"
input_player1_analog_dpad_mode = "1"
```

HerbFargus, 2016-01-19:
its `/opt/retropie/configs/all/retroarch-joypads`

joolswills, 2016-01-19:
thanks. yeh I'm very tired. nite - herb can take over :)

johnjelinek, 2016-01-19:
@HerbFargus: I never switched it to `udev`, it was always set to `udev`.

johnjelinek, 2016-01-19:
This is all that's in `retroarch-joypads`:
```
pi@raspberry:/opt/retropie/configs/all/retroarch-joypads $ cat MicrosoftX-Box360pad.cfg

input_device = "Microsoft X-Box 360 pad"
input_driver = "udev"
input_r_y_plus_axis = "+4"
input_l3_btn = "9"
input_r_x_minus_axis = "-3"
input_l2_btn = "4"
input_start_btn = "7"
input_exit_emulator_btn = "7"
input_l_y_minus_axis = "-1"
input_l_axis = "+2"
input_load_state_axis = "+2"
input_up_btn = "h0up"
input_r_y_minus_axis = "-4"
input_a_btn = "0"
input_b_btn = "1"
input_reset_btn = "1"
input_down_btn = "h0down"
input_r2_btn = "5"
input_l_x_plus_axis = "+0"
input_l_y_plus_axis = "+1"
input_r3_btn = "10"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_x_btn = "2"
input_menu_toggle_btn = "2"
input_select_btn = "6"
input_enable_hotkey_btn = "6"
input_l_x_minus_axis = "-0"
input_y_btn = "3"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_r_x_plus_axis = "+3"
```

HerbFargus, 2016-01-19:
I'm not sure what the issue with it could be as the configs look valid.  We should be releasing a new image soon and it might be worth testing on that just to make sure that there aren't any previous conflicting configurations.

johnjelinek, 2016-01-19:
how do I downgrade my ES w/ an older binary so I can have my controller working again? (I'm assuming that might be faster than waiting for a new image)

HerbFargus, 2016-01-19:
Ive sent you a link to your email.

Emulationstation isn't the issue as your controller works fine in it, rather retroarch isn't reading your inputs for some reason.

johnjelinek, 2016-01-19:
@HerbFargus: I just realized that the controller in the emulator is actually mapping to `port #1`, so -- controller 2. Which is why buttons aren't working. But I only have 1 controller. Anyway to enforce it to go to `port #0`?

johnjelinek, 2016-01-19:
I just installed the new image. It would be pretty sweet if future releases also included the steam controller driver, so I don't have to get that manually: https://github.com/ynsta/steamcontroller

HerbFargus, 2016-01-19:
You can manually edit:

```
# If desired, it is possible to override which joypads are being used for user 1 through 8.
# First joypad available is 0.
# input_player1_joypad_index = 0
# input_player2_joypad_index = 1
```

So you'd change 

input_player1_joypad_index = 0

To 

input_player1_joypad_index = 1

johnjelinek, 2016-01-19:
dang .. even after imaging, if I start an emulator it says `No xbox controller configured` followed by `configured on port #1`. So only 2nd player works.

johnjelinek, 2016-01-19:
I did what you suggested `input_player1_joypad_index = 1` and now my buttons are mapped real weird. `start button` resets the rom, `X` brings up the emulator menu, nothing like what I'm expecting.

johnjelinek, 2016-01-19:
well, d-pad works though, but `a`,`b`,`x`,`y` don't work as expected.

johnjelinek, 2016-01-20:
Are there any other options I can do to troubleshoot?

johnjelinek, 2016-01-21:
ahh -- I figured it out. By having the xboxdrv enabled, ES was recognizing 2 controllers plugged in. When the steam controller is emulating an xbox controller, make sure xboxdrv is disabled.

joolswills, 2016-01-21:
I think we can close this now. the config generation in the OP i solved also. cheers.

closed 2016-01-21 by joolswills