RetroPie/RetroPie-Setup #1181: Bluetooth daemon fails to start

PeterPablo, 2016-01-10:
I am running a raspbian jessie with a recently cloned retropie setup script. During boot my bluetooth daemon fails to start:

```
root@rpi2:~# journalctl -xn
-- Logs begin at Sa 2016-01-09 15:47:13 CET, end at So 2016-01-10 10:49:31 CET. --
Jan 10 10:49:24 rpi2 su[11048]: + /dev/pts/0 root:root
Jan 10 10:49:24 rpi2 su[11048]: pam_unix(su:session): session opened for user root by pi(uid=0)
Jan 10 10:49:31 rpi2 sudo[11074]: root : TTY=pts/0 ; PWD=/root ; USER=root ; COMMAND=/usr/sbin/service bluetooth start
Jan 10 10:49:31 rpi2 sudo[11074]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Jan 10 10:49:31 rpi2 systemd[1]: Starting Bluetooth service...
-- Subject: Unit bluetooth.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit bluetooth.service has begun starting up.
Jan 10 10:49:31 rpi2 systemd[11098]: Failed at step EXEC spawning /usr/lib/bluetooth/bluetoothd: Permission denied
-- Subject: Process /usr/lib/bluetooth/bluetoothd could not be executed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- The process /usr/lib/bluetooth/bluetoothd could not be executed and failed.
-- 
-- The error number returned while executing this process is 13.
Jan 10 10:49:31 rpi2 systemd[1]: bluetooth.service: main process exited, code=exited, status=203/EXEC
Jan 10 10:49:31 rpi2 systemd[1]: Failed to start Bluetooth service.
-- Subject: Unit bluetooth.service has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit bluetooth.service has failed.
-- 
-- The result is failed.
```

I think this is due to the execute flag being removed [here](https://github.com/RetroPie/sixad/blob/master/sixad#L67): `$SUDO chmod 644 "$BLUETOOTHD"`.

What is the reasoning behind modifying the rights of the bluetooth daemon?

Is my raspbian wrong in trying to automatically start the bluetooth daemon? Is it intented that `sixad` takes care of starting `bluetoothd`? I am not familiar with the code but wanted to point out my issue.

I am trying to use a PS3 controller. No success yet...

gizmo98, 2016-01-10:
Sixad replaces bluetoothd. Both daemons cannot run at the same time. Sorry, this is intended.

gizmo98, 2016-01-10:
Do you use an original PS3 controller?

PeterPablo, 2016-01-10:
Ah, ok! I did not know that. So in order to connect my original PS3 controller (yes) I need to ensure that the bluetooth daemon is stopped and the sixad service is running?
I know that I once saw the device under `/dev/input/js0`, but `jstest` didn't show changing values when pressing buttons.

I will be able to conduct further tests tomorrow. Should I first Du some test when the controller is connected using USB? Should jstest there show me chaning values? What log message should I look for, to determine if the connection was established successfully?

Thank you for your help!

gizmo98, 2016-01-10:
If you ps3 controller is connected over usb you must press PS button once to enable button input.

PeterPablo, 2016-01-10:
Thanks. I'll retest tomorrow and report back.

PeterPablo, 2016-01-11:
After connecting the controller using USB I get the following in `syslog`:
```
Jan 11 19:36:26 rpi2 kernel: [  934.895312] usb 1-1.5.2: new full-speed USB device number 9 using dwc_otg
Jan 11 19:36:26 rpi2 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Jan 11 19:36:56 2016 [try http://www.rsyslog.com/e/2007 ]
Jan 11 19:36:26 rpi2 kernel: [  935.017656] usb 1-1.5.2: New USB device found, idVendor=054c, idProduct=0268
Jan 11 19:36:26 rpi2 kernel: [  935.017694] usb 1-1.5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jan 11 19:36:26 rpi2 kernel: [  935.017712] usb 1-1.5.2: Product: PLAYSTATION(R)3 Controller
Jan 11 19:36:26 rpi2 kernel: [  935.017728] usb 1-1.5.2: Manufacturer: Sony
Jan 11 19:36:26 rpi2 systemd-udevd[2278]: failed to execute '/lib/udev/mtp-probe' 'mtp-probe /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.2 1 9': No such file or directory
Jan 11 19:36:27 rpi2 kernel: [  935.315299] usb 1-1.5.2: reset full-speed USB device number 9 using dwc_otg
Jan 11 19:36:27 rpi2 kernel: [  935.514476] input: Sony PLAYSTATION(R)3 Controller as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.2/1-1.5.2:1.0/0003:054C:0268.0005/input/input2
Jan 11 19:36:27 rpi2 kernel: [  935.515943] sony 0003:054C:0268.0005: input,hiddev0,hidraw3: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-3f980000.usb-1.5.2/input0
Jan 11 19:36:27 rpi2 upowerd[1466]: (upowerd:1466): UPower-Linux-WARNING **: no valid voltage value found for device /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.2/1-1.5.2:1.0/0003:054C:0268.0005/power_supply/sony_controller_battery_00:01:4a:9b:b9:9c, assuming 10V
```

Afterwards I can do `jstest /dev/input/js0` but the output is stale and doesn't react to key presses.
After pressing the PS button and retrying with `jstest` it still does not change output.

Could it be the failing `mtp-probe`?

PeterPablo, 2016-01-11:
Disconnecting from USB and again pressing the PS button yields:
```
Jan 11 19:42:55 rpi2 kernel: [ 1323.282645] usb 1-1.5.2: USB disconnect, device number 9
Jan 11 19:42:55 rpi2 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Jan 11 19:43:55 2016 [try http://www.rsyslog.com/e/2007 ]
Jan 11 19:43:36 rpi2 sixad-bin[1382]: unable to connect to sdp session
Jan 11 19:44:16 rpi2 sixad-bin[1382]: unable to connect to sdp session
Jan 11 19:44:16 rpi2 sixad-bin[1382]: HID create error 110 (Connection timed out)
Jan 11 19:44:16 rpi2 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Jan 11 19:45:16 2016 [try http://www.rsyslog.com/e/2007 ]
Jan 11 19:44:16 rpi2 kernel: [ 1404.245949] Bluetooth: hci0 link tx timeout
Jan 11 19:44:16 rpi2 kernel: [ 1404.245992] Bluetooth: hci0 killing stalled connection 00:01:4a:9b:b9:9c
Jan 11 19:44:16 rpi2 kernel: [ 1404.246014] Bluetooth: hci0 link tx timeout
Jan 11 19:44:16 rpi2 kernel: [ 1404.246023] Bluetooth: hci0 killing stalled connection 00:01:4a:9b:b9:9c
```

PeterPablo, 2016-01-11:
Tail of the `sixad` log:
```
Starting bluetooth (via systemctl): bluetooth.service.
sixad-bin[1387]: started
sixad-bin[1387]: sixad started, press the PS button now
sixad-bin[1387]: Done
Starting bluetooth (via systemctl): bluetooth.service.
sixad-bin[1382]: started
sixad-bin[1382]: sixad started, press the PS button now
sixad-bin[1382]: unable to connect to sdp session
sixad-bin[1382]: unable to connect to sdp session
sixad-bin[1382]: HID create error 110 (Connection timed out)
sixad-bin[1382]: unable to connect to sdp session
sixad-bin[1382]: HID create error 107 (Transport endpoint is not connected)
sixad-bin[1382]: unable to connect to sdp session
```

gizmo98, 2016-01-11:
The upower-linux-warning message sounds strange. Is your power supply good enough?

PeterPablo, 2016-01-17:
I had tried it again with an active USB hub, but it didn't change things. I'd like to reiterate my two earlier questions:
* Could it be the failing mtp-probe?
* Should `jstest` show changing values, when my PS3 controller is simply connected via USB?

joolswills, 2016-01-20:
I don't think jstest will work when connected but not 100%.

Please try with a fresh 3.4 image from https://github.com/RetroPie/RetroPie-Setup/releases/tag/3.4 - to rule out something with your current software stack.

PeterPablo, 2016-01-20:
Please give me some time to do the requested test. I will first have to find an unused sd card :-) Thank you for your ongoing initiative!

closed 2016-03-21 by joolswills

joolswills, 2018-07-10:
Please use our forum for support. This ticket has been closed.