RetroPie/RetroPie-Setup #1182: USB gamepad needs reconnect to be usable

speters, 2016-01-10:
My USB gamepad needs to be reconnected (plug off, plugged in) after choosing an emu/game in Emulationstation. Without doing this, it can not be used inside the game.
In Emulationstation, gamepad works fine without needing to re-plug it into the usb port.

Gamepad type is Logitech Gamepad F310. It is configured with `input_driver = udev`.
[Output of lsusb -v -d 046d:c21d](https://github.com/RetroPie/RetroPie-Setup/files/84099/lsusb_logitechf310.txt)

The behaviour is the same on Raspberry Pi B (old 256MB model), and Raspberry Pi2, both running Retropie 3.3 from the official images.

**dmesg:**
```
[    6.072633] input: Logitech Gamepad F310 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.2/1-1.3.2:1.0/input/input3
[    6.075716] usbcore: registered new interface driver xpad
```

lsmod shows - amongst others - these modules loaded: `xpad, ff_memless, joydev, uinput, evdev`
Devices are rw for group input, user pi is member of this group.

Looking forward to get some suggestions on how to solve this,
with kind regards from Hamburg, Germany,
Sönke

HerbFargus, 2016-01-10:
Known issue: 

Only current solutions:

- downgrade kernel
- tap top right shoulder button to get it to be recognised
- skip configuring top left and top right shoulder altogether

See video if you have issues: https://youtube.com/watch?v=AhkEnDdygbQ

speters, 2016-01-12:
Thanks for your reply.
But I think you missunderstood, my initial controller config has already been done successfully. First automatically in Emulationstation (overcoming the shoulder button misbehaviour like you also showed in your video), then manually added the hotkey button in the config files.

When starting a SNES game, it shows controller configured just after the text mode messages:
[image]
In this stage, controller is not usable.

You mentioned a kernel downgrade as a possible solution. Did you mean xpad to older xboxdrv "downgrade" or a real regression in the kernel?

closed 2016-01-17 by joolswills