RetroPie/RetroPie-Setup #1184: How do I know if I have the latest version?

smartroad, 2016-01-12:
Hi not really an issue, but is there anywhere on the script that lets you know you are using the latest version? Just updated the script as I noticed I missed 3.2 and 3.3 but although it says it has updated there is no obvious change (obviously(!) as the menu options remain the same :grin:)

joolswills, 2016-01-12:
You can get help on the support forum - if you have updated retropie-setup script and have done a full binary install, you will have the very latest retropie-setup code (post 3.3), with binaries that are mostly 3.3 with a few updates.

info regarding version here - but it's not as simple as haivng a v3.2 / v3.3 those are reallly just used to the images.

https://github.com/RetroPie/RetroPie-Setup/issues/949

https://github.com/RetroPie/RetroPie-Setup/wiki/Updating-RetroPie

closed 2016-01-12 by joolswills

craigerskine, 2016-01-15:
It would be nice to know if an update is needed. Like if there is a new emulator or script or something else is out of date. Then we would know if we are running the latest versions of everything.

joolswills, 2016-01-15:
There are plan to improve the way things are installed and updated - but nothing to report yet.