RetroPie/RetroPie-Setup #1185: Bluetooth commands no longer work on Jesse

teirich, 2016-01-13:
Trying out the Jesse image but I've been unable to connect Bluetooth controllers due to all the instructions for Bluez being outdated https://github.com/RetroPie/RetroPie-setup/wiki/Setting-up-a-Bluetooth-controller

joolswills, 2016-01-14:
Yeah, the wiki needs updating for Jessie, and the bluetooth gui in retropie also.

in the meantime you can pair from terminal - see the CLI instructions on this page - https://wiki.debian.org/BluetoothUser

Feel free to add Jessie specific information to the wiki if you get it sorted.

teirich, 2016-01-15:
Yea unfortunately with Bluez 5.x being added in Jessie, it looks like bluetoothctl is the only command-line tool supported.  This is making scripting connecting on startup & other tasks  really painful as it uses an interactive shell.  I'm able to get a controller paired using `sudo bluetoothclt` (sudo for good measure) & the following commands 

>power on
>scan on  //you should start seeing devices populate here, take note of the MAC address of the device you want to use
>agent on 
>pair <MAC Address>
>trust <MAC Address>
>connect <MAC Address>  (I'm using an 8Bitdo SFC30 GamePad and it seems to timeout relatively quickly.  Usually I turn it off/on again to for this step to ensure it hasn't timed out)
>exit

In order to get it to register as a Joypad I had to add a udev rule `sudo vim /etc/udev/rules.d/10-local.rules`  
>SUBSYSTEM=="input", ATTRS{name}=="8Bitdo SFC30 GamePad Joystick", MODE="0666", ENV{ID_INPUT_JOYSTICK}="1"

I'm working on scripting something to run on startup and I'm still having trouble getting the D-Pad to map in EmulationStation.  I'll make a new section if I get everything working like I want, in the meantime I'll leave this here.

joolswills, 2016-01-17:
the bluetooth module in retropie has now been updated to work on jessie. tested only on a bluetooth keyboard here. a combination of bluetoothctl + bluez5 updated python test scripts. Might come up with another solution later.

some controllers may not connect automatically on startup, but my keyboard did and worked fine.

teirich, 2016-01-19:
Thanks, this did fix my problem. Working like a charm.

closed 2016-01-19 by teirich

joolswills, 2016-01-19:
did it pair on reboot automatically, or did you have to get it to connect each time ? Did you need the udev rule also ?

teirich, 2016-01-19:
Yea it's been pairing automatically, no problems.  I did need the udev rule to make it recognize the controller as a joypad. Not sure if that's to do with the controller itself, or if that's the case for any Bluetooth joypad. Maybe its not necessary if you use the sdl driver instead.

joolswills, 2016-01-19:
awesome thanks. Does it make a difference how you power it on - I believe those joypads have different power on modes etc ?

teirich, 2016-01-19:
Yes, the only 'mode' i was successful with was Joystick mode.  For SFC30 that's START+R when you power it on.  When you discover the device via bluetooth, it should look something like '8Bitdo SFC30 GamePad Joystick' with 'Joystick' at the end of the name.  If I used the default mode (just holding START at startup) I wasn't able to get the DPad to work in emulationstation/retroarch and I didn't have any luck with the other mdoes.

joolswills, 2016-01-19:
Thanks for the clarification. useful to know (and for other users if they come across this ticket).

biscuits99, 2016-01-23:
@teirich I've only tried the FC30 Pro, but am about to give the SFC30 a go. To confirm, did you update retropie script, run the bluetooth setup and it paired, connected and automatically re-connected on reboot without issue? (Albeit with the udev rule).
I'm going to try a couple of different bluetooth receivers in case thats where my issue is.

biscuits99, 2016-01-23:
@teirich Ok, SFC30 works well as above, I used it with firmware 2.65. Although I'm pretty sure the config files dont have the right button mappings. I'll open that issue separately.

teirich, 2016-01-25:
@biscuits99 Hey, you are correct.  I believe emulationstation is mapping the buttons using sdl driver, but retroarch is using udev.  Just go into the RetroPie menu, launch RetroArch GUI and re-map your buttons there.