RetroPie/RetroPie-Setup #1187: Steam Controller support

johnjelinek, 2016-01-16:
I tested this driver on the raspberry pi and it works great: https://github.com/ynsta/steamcontroller. However, it currently is set up to emulate an xbox controller since ES doesn't recognize steam controller as its own type. I'd like to request a steam controller as its own type to have even better control. For example, I'd like the right pad to act like a mouse except when pushed in, then I'd like it to be like my right analog.

joolswills, 2016-01-20:
We don't yet support it but may in future.

johnjelinek, 2016-02-07:
@joolswills: if I wanted to hack on adding this support using https://github.com/cvuchener/steamcontroller-linux-kernel as a generic controller driver, where should I start? I suppose changes in ES and in RetroArch?

johnjelinek, 2016-02-07:
I suppose the driver ^ could be remapped with xboxdrv and then it might work without additional changes to ES/RetroArch. What do you recommend?

Whovian9369, 2016-03-26:
Going to +1 this myself, since this bloody thing is seen as only a keyboard, and I would want it to work as more like a 360 controller, or as its own controller. (As John has said in the original issue.) 

I love the controller but both RetroPie and Lakka ( http://www.lakka.tv/ ) don't use it as a controller, and that's oddly pissing me off.

Thanks for the software! I'll definitely keep following it!

johnjelinek, 2016-03-28:
There is a userland driver and a kernel module driver available, but both
have their quirks at the moment. It'd definitely be nice to get some more
momentum on this support.

Twitch0815, 2016-04-06:
Looking at this but it looks concerning cause one user reports such high cpu usage, will try it out and report back. https://github.com/ynsta/steamcontroller

johnjelinek, 2016-04-06:
The CPU has been fine on my Pi3

veleiro, 2016-05-15:
+1
Something funky with the buttons sending multiple presses when one is pressed as well.

johnjelinek, 2016-06-04:
I just bought a 2nd steam controller. The driver mentioned at the top doesn't currently support multiple connected controllers. The pi sees them both though:
```
$ lsusb -d 28de:1142
Bus 001 Device 006: ID 28de:1142
Bus 001 Device 005: ID 28de:1142
```
I think it's time to get better direct support for these controllers. What's the best way to proceed? Contributing to the python userland driver (which runs on `libusb`) or a kernel module (but pi's don't really come w/ hackable kernels and the kernel headers are a pain to set up)?

johnjelinek, 2016-06-04:
on-topic: [Steam Controllers](http://store.steampowered.com/app/353370/) are on sale this weekend, in case that encourages further development on this :)

XD3l, 2016-06-04:
I ordered one this morning! :)

“Therefore do not deceive yourself! Of all deceivers fear most yourself!”
― Søren Kierkegaard
<https://www.goodreads.com/author/show/6172.S_ren_Kierkegaard>

CoolJoe72, 2016-08-08:
I just picked up a Steam controller (Doom bundle sale), I'm more than willing to help out with testing and stuff. Where do we get started?

cy2k, 2016-08-14:
I'm new to RPi and RetroPie, but man this would be sweet, so I'd love to add my support to getting this functionality and help out in any way I can.

Wolf-Fivousix, 2016-08-22:
Just set up a RetroPie myself but got kind of bummed it doesn't accept the Steam Controller by default, as it is now my favorite controller by far.

joolswills, 2016-08-22:
@Wolf-Fivousix @cy2k This is a issue on the tracker for tracking steam support - please keep the discussion to things that help reach the aim of the ticket. Github has reactions if you want to add a +1 etc.

Wolf-Fivousix, 2016-08-22:
@joolswills Thank you, I am new to this. Found the +1 and did it. I'll delete my post (as well as this one).

joolswills, 2016-08-23:
I will implement the existing driver as linked from the first post (I have written most of the retropie module, just need to do some testing). I don't actually have a steam controller, so it would be useful if someone can test the module for me when it's done.

It will do until an alternative driver comes along.

Wolf-Fivousix, 2016-08-23:
@joolswills , If you do that for us, I'll be happy to help you out with the testing. I am not very familiar with Linux, but I think that wouldn't be a problem.

Whovian9369, 2016-08-24:
It would give me something to do with my rPi2, so I'll happily try and help test!

jlmassey, 2016-09-20:
my self would love for this to come to life.

skorokithakis, 2016-10-14:
@joolswills I use the Steam controller pretty much exclusively as well, so I would be happy to test. Without Steam controller support, there's no useful way for me to control RetroPie :(

bruinfan0707, 2016-10-15:
@joolswills I am fairly new, but would be willing to help as well. I purchased two of these controllers just incase you need help testing multiple controllers.

joolswills, 2016-10-15:
Thanks - I have created a PR for it here - https://github.com/RetroPie/RetroPie-Setup/pull/1718

Ideally for testing it would be someone who is comfortable with the linux commandline

to switch to the branch -

```
cd ~/RetroPie-Setup
git checkout -b joolswills-steamcontroller master
git pull https://github.com/joolswills/RetroPie-Setup.git steamcontroller
```

and then to switch back to the main RetroPie branch

```
git checkout master
```

bruinfan0707, 2016-10-15:
@joolswills I may leave this up to the other posters. I entered the commandlines given and the control did not register. I am pretty new to Linux and I am very much still available to help, but I don't want this to be an uphill battle (teaching a noob the basics).

Things I tried: Entered the steam branch and rebooted pi. Tried to register controller on configure input menu, but no gamepads detected.

2nd attempt- Entered both commandlines steam branch and retropie branch immediately after. Same message on input menu. 

Thanks again for all the help J

HerbFargus, 2016-10-15:
Did you enable it in the setup script? The commands were only to get the setup script with the latest module, you have to enable the driver through the setup script like you would with any other specialty controller

bruinfan0707, 2016-10-15:
@HerbFargus @joolswills Thanks Herb! That worked. Mapped the controller on Xbox 360 mode and is working perfectly. Need to run some errands so I won't be able to try desktop mouse/keyboard mode till later. Thank you @joolswills for all you do and thanks for the quick fix @HerbFargus. 

P.S Any chance this will ever work with two steam controllers? It still only recognizes one despite both being plugged in by separate dongles.

HerbFargus, 2016-10-15:
Seems it is being looked at: https://github.com/ynsta/steamcontroller/issues/25

skorokithakis, 2016-10-16:
Works great for me as well. The keys are a bit fiddly to map, and RetroPie uses the d-pad while Kodi uses the analog stick, but it's good enough on the driver side.

joolswills, 2016-10-16:
Thanks for the feedback. Will merge.

ifoundgoldbug, 2016-10-31:
Thank you very much for for this, i bought a pair of steam controllers in that sale in june and did not have the technical skill to get them to function wired or wirelessly. You have given me hope to try again and I would gladly do more testing for you on the controls and multiple controllers.

joolswills, 2016-10-31:
The code is in place now - so you can update retropie-setup and give it a go. Thanks - closing this issue now.

closed 2016-10-31 by joolswills

brugnara, 2016-11-13:
@joolswills can we please have some instructions?

jeremycg, 2016-11-24:
@brugnara Thanks for those instructions!

To save some time for anyone following along, the step 4 above is in the setup at:

Manage Packages > Manage Driver Packages > 826 Steam Controller.

NowayIndustries, 2016-11-26:
Just a heads up, i just spent a couple of hours trying to get this to work:
I needed to install 'python-dev' and 'python3-dev' before it would finish the script install inside of manage drivers (it would not give an error but the log had it).
Inside the log you could find errors about Python.h no such file or folder, when it attempted to install psutil.
Other then the twitchyness of actually mapping inputs via emulationstation it was a pretty much perfect.
This was using a 3B, with 4.1 written to the sdcard, a fresh copy.
So to add to @brugnara steps:
1.5.1 sudo apt update
1.5.2 sudo apt install python-dev python3-dev

joolswills, 2016-11-26:
Thanks. I did a test and it didn't require python-dev here, just python3-dev - but I have added that dependency to the script.

hheimbuerger, 2016-12-06:
I've started from a blank slate RetroPie 4.1 (first timer here) and only activated the Steam Controller driver. I had the same issue of an `ImportError` on `psutil` during boot. (Even after installing `python-dev` and `python3-dev` via apt.)

Instead, I solved it by installing `psutil` into the virtualenv that the driver is using, roughly like this:

```
/opt/retropie/supplementary/steamcontroller/bin/pip install psutil
```

joolswills, 2016-12-06:
@hheimbuerger This issue tracker is not used for support btw (and this ticket is closed) - please use Fargusherb

I will re-test from a fresh install, but the fix was done after 4.1 so you would need to update retropie-setup first.

joolswills, 2016-12-06:
Will check the psutil error.

hheimbuerger, 2016-12-06:
@joolswills Sure, sorry for that, I removed my unrelated question. This was the only professional conversation I could find about RetroPie's Steam Controller integration, as there's no wiki page. So that's why I posted it here. But you're right that it's off-topic here, so I deleted it.

joolswills, 2016-12-06:
Sorry. clipboard failure - https://retropie.org.uk/forum/ not what I wrote before.

fairyforest, 2016-12-18:
Hey everybody. After a lot of shipping problems and costs, I finally received my controller and I am crazy to use it on RetroPie. I found the driver and got it installed but I can't get it running, nor cabled nor wireless. I have a PS4 Controller configured, I guess I have to switch the controllers somehow.
I also sign up for testing purposes.
It would be really great to get something like antimicro, qjoypad or xpadder (this one is for Windows) working with the Steam Controller to break up its full potential.
Enjoy.

LeDucDuBleuet, 2017-11-17:
I was able to have two Steam controllers working well with RetroPie after installing the required package and pairing them both to the same dongle with the Steam software on my Ubuntu desktop.

You also have to modify your **/etc/rc.local** by replacing the line:
`"/opt/retropie/supplementary/steamcontroller/bin/sc-xbox.py" start`

By these two lines:
```
"/opt/retropie/supplementary/steamcontroller/bin/sc-xbox.py" start -i0
"/opt/retropie/supplementary/steamcontroller/bin/sc-xbox.py" start -i1
```
Based on: https://github.com/ynsta/steamcontroller/issues/25

purpleidea, 2018-01-11:
So seems to be a few things: (1) udev rules with proper *group* Group is super important or it won't work. Then *needs* an unplug/replug. (2) ./sc-xbox.py start (somehow) (3) key mapping: https://gist.github.com/purpleidea/81056462c8c780fd03d2012071b139a7 … Gah!

HTH