RetroPie/RetroPie-Setup #1192: Kodi: no subtitles displayed

havnar, 2016-01-18:
I have tried some .mkv files but there is nothing being displayed in regard of subtitles.
The UI says it is enabled but nothing is rendered.

Is this a bug or just a missing feature for now?

robertybob, 2016-01-18:
Not a RetroPie issue (?) have you looked here? http://kodi.wiki/view/Subtitles

havnar, 2016-01-18:
This relates to the retropie package of Kodi of course, and yes I have looked there, subtitles should be enabled but are not displayed.
Kept the settings vanilla.

zerojay, 2016-01-18:
You'll probably have to ask the Kodi upstream developers about that. Perhaps it does not support your form of embedded subtitles.

joolswills, 2016-01-18:
we use the kodi packages shipped with raspbian, so nothing we can do sorry.

closed 2016-01-18 by joolswills