RetroPie/RetroPie-Setup #1193: ps3 controller timeout issue

daweze, 2016-01-18:
clean install of build 3.3.1 making no changes but to update retropie-setup, install/pair the ps3 controller an add to sixad default profile.

enable_timeout 1
timeout_mins 15 

The controller will not timeout and turn off this worked in 3.2.1

gizmo98, 2016-01-18:
Could be the new driver does not support timeout anymore.

bengus64, 2016-01-23:
daweze: we both created similar threads on the retropie forum. I just wanted to chime in that I recently updated to 3.4, completely removed sixad and re-installed. Unfortunately I still have the same problem as you.

enable_timeout 1
timeout_mins 15 (or any value)

Still no longer works.

daweze, 2016-01-23:
bengus64 i did the same with 3.4 today and like you still have the same problem.  i also tried working with /etc/bluetooth/input.conf IdleTimeout=XX and got no where with that either.

gizmo98, 2016-01-23:
Sorry guys. I now there is room for enhancements. If you came up with any solution i can add it. You could also try to enable this option "enable_safety_timeout = 1".

For now i have to sort out how to integrate an install option for a shanwan branch and multiple gasia controller support. Some of you posted a link to a wotking shanwan repo. It is also possible our new raspbian jessie based image's bluetooth daemon has builtin ps3 controller support. If this is true we could remove sixad at all.

joolswills, 2016-01-23:
I can confirm pairing with jessie bluez does work. Although bluetoothctl kept spamming output during the process. 

I basically did the same as https://wiki.gentoo.org/wiki/Sony_DualShock

joolswills, 2016-01-24:
I would add it was somewhat fiddly - and I had to try a few times to get it paired. Seemed to be better on the more recent Bluez on Ubuntu 15.10

Perhaps we can do a combination of using sixpair tool + bluez on jessie.

gizmo98, 2016-01-25:
The new bluez should have builtin sixpair. I don't know if it is always triggered if you connect a controller over usb. But we could do something like sixpair + bluez (if clone support is even better).

joolswills, 2016-01-25:
The latest bluez on ubuntu pairs correctly and picks up the controller when plugged via usb. the one with jessie doesn't seem as smart. I had to do "pair XX" any times before I got an auth request etc.

Lakka refers to some of the problems;
http://www.lakka.tv/doc/Wireless-Dualshock/

joolswills, 2016-01-25:
I think sixpair + bluez would def be worth looking into

gizmo98, 2016-03-04:
@daweze @bengus64 
Controller timeout is back. Please remove and reinstall controller driver. https://github.com/RetroPie/sixad/commit/99176054ad3d9c4e8f7ba1cb82eca1ed012c78c9

bengus64, 2016-03-04:
gizmo98 thank you so much for all of your hard work and time, I really appreciate it. I just wanted to confirm that this is working for me perfectly now (Retropie 3.6 build, RP2). Thank you again!

gizmo98, 2016-03-06:
Thank you. Main thanks go to @Dark-Show because he contributed this patch.

closed 2016-03-06 by gizmo98