RetroPie/RetroPie-Setup #1194: Add Love Game Engine Module

HerbFargus, 2016-01-19:
Just posting for discussion on adding the love game engine to play games like Mari0:

from this post here:

http://blog.petrockblock.com/forums/topic/adding-mari0-and-sunvox-modular-music-studio/

May also be worth looking at a RetroArch option:

https://github.com/libretro/libretro-lutro

petrockblog, 2016-01-19:
I have also followed that thread and also the idea of maybe integrating the Love framework into RetroPie recently.

I find that very interesting and, actually, this could become something like a new aspect of RetroPie IMO: A platform to access open-source game projects. This might not only make use of the Love framework, but maybe also on other frameworks ... But that would definitely be a long shot.
The question during the last podcast interview for a vision for RetroPie made me think about it. Maybe this is something ...

Is the RetroArch option working well and continuously developed? Or does building the original Love sources seem to be the better way? I still need to look into both more closely ...

HerbFargus, 2016-01-19:
I haven't tested the retroarch build yet as its still experimental so I don't even know if it compiles, Though typically as we've seen, performance on the original emulators and engines often have better performance than their retroarch equivalents. Though it may be worth trying both out.

Another obstacle is that the Love framework changes with each point release and renders the former game versions often unplayable unless there is still active development on the game, but it seems the lua development community is quite active.

HerbFargus, 2016-01-19:
Just a preliminary Module for reference: Still haven't tested it and it will likely need changes:

http://pastebin.com/qwyKRD33

joolswills, 2016-01-19:
depends_love

build_love

install_love

*sniff*

so romantic. :)

HerbFargus, 2016-01-19:
if only it were that easy. 

I'm also unsure if there are build differences between wheezy and Jessie, I modeled the module from jeffdaman's wheezy build- I'll have to test it with both.,

HerbFargus, 2016-01-20:
So from the former build I got this build log: http://pastebin.com/NyHG0ZUw

So after changing a few things the following works. Its a bit rudimentary. I'm thinking future-wise that it would be good to build in a game selector (like we are thinking of doing for wads with doom)

```
#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

rp_module_id="love"
rp_module_desc="Love2d Game Development Framework"
rp_module_menus="4+"
rp_module_flags="!mali"

function depends_love() {
    getDepends build-essential automake libtool libphysfs-dev libsdl1.2-dev libopenal-dev liblua5.1-0-dev libdevil-dev libmodplug-dev libfreetype6-dev libmpg123-dev libvorbis-dev libmng-dev libxpm-dev libxcursor-dev libxxf86vm-dev mercurial
}

function sources_love() {
    hg clone https://bitbucket.org/rude/love "$md_build"
}

function build_love() {
    cd /root/love
    ./platform/unix/automagic
    ./configure --with-lua=lua
    make clean
    make
}

function install_love() {
    make install 
}

function configure_love() {
    mkRomDir "ports"
    mkRomDir "ports/love"

    # get Mari0 10.0 (freeware game data)
    wget -q -O mari0.love https://github.com/radgeRayden/future-mari0/releases/download/v0.1/mari0.love
    mv mari0.love "$romdir/ports/love/"

    addPort "$md_id" "love" "Love" "love $romdir/ports/love/mari0.love"
}
```

joolswills, 2016-01-20:
Ta daa https://github.com/RetroPie/RetroPie-Setup/commit/6b752d1f5cb162e96f519e74b26e70a803a94bae

Thanks for getting this started Herb.

HerbFargus, 2016-01-20:
Thanks @joolswills looks brilliant. I give it another test when I get a chance but will consider this closed. Will reopen if I encounter any issues.

closed 2016-01-20 by HerbFargus