RetroPie/RetroPie-Setup #1195: lr-mednafen-pce missing from Retropie 3.4?

davidj1987, 2016-01-20:
Hey,

I found that some PCE/TG16 games work better, and Supergrafx games work in "lr-mednafen-pce" plain not the fast one. The "lr-mednafen-pce-fast" core works.

I updated to 3.4 and found that it's not working. I tried looking in the install script and cannot find it so I cannot reinstall from the update script. Just curious if it was intentional, but again the pce-fast core still works.

HerbFargus, 2016-01-20:
mednafen-pce doesnt exist anymore. it was superceded by lr-mednafen-pce-fast

https://github.com/RetroPie/RetroPie-Setup/issues/1088

davidj1987, 2016-01-20:
Hmm... there are some advantages to the non-fast core. By any chance is it possible that Beetle/Mednafen SGX could be ported to Retropie?

joolswills, 2016-01-20:
We could certainly try it.

HerbFargus, 2016-01-20:
@joolswills

Is this what we essentially removed? 

https://github.com/libretro/beetle-supergrafx-libretro

joolswills, 2016-01-20:
well not according to one of their devs, but perhaps they did rename it or they did a better port. I can add it and see.

davidj1987, 2016-01-20:
Thanks HerbFargus for referencing the past issue and joolswills looking into this. 

The SuperGrafx games that ran in the older mednafen-pce were fullspeed and no issues, a small library sure

HerbFargus, 2016-01-20:
7 games from what I understand but also supposedly backwards compatible with all of the previous games which makes it more like hundreds. (Unless of course this port only provides support for those 7 games)

davidj1987, 2016-01-20:
The supergrafx port, on Windows x64 (only version I tried) also runs PC Engine and PC Engine CD games. The Supergrafx was just a beefed up PC Engine and played all the games on hucard and CD.

joolswills, 2016-01-20:
i've built it. will test and add if it works.

joolswills, 2016-01-20:
added - if you update retropie-setup you can find it in experimental (to build from source)

davidj1987, 2016-01-20:
I get an error saying could not successfully build as "mednafen_supergrafx_libretro.so" could not found.

joolswills, 2016-01-20:
I would need to see a build log from logs/. Probably out of memory or something - I didn't test on wheezy either.

davidj1987, 2016-01-20:
As requested

[rps_2016-01-20_174139.txt](https://github.com/RetroPie/RetroPie-Setup/files/97832/rps_2016-01-20_174139.txt)

joolswills, 2016-01-20:
looks like it needs a modification to add "-lrt" to the link libs. This wasn't an issue on Jessie - assuming you are on wheezy, I would get a new 3.4 Jessie image anyway, as wheezy is not going to be supported in the future.

davidj1987, 2016-01-21:
Did a fresh install from a 3.4 Jessie image, and it works great with the
SuperGrafx games!

davidj1987, 2016-01-31:
Thanks again for adding the core. If I could make one more suggestion, IMO I would keep both the Supergrafx core (if it gets out of experimental) and the PCE Fast Core. 

For example, Space Harrier, both the US and JP versions do not run on the Supergrafx core as they will crash. Turns out this happens on real hardware when it is in Supergrafx mode. What you can also do is use System Card 3.0 with the Supergrafx Core and use a lesser version of the system card with the PCE Fast Core. No idea if other games have issues though.

I know the CD version of Altered Beast (again on real hardware) only works on System Card 1.0 and other games have easter eggs if you have a lesser system card than 3.0 like Dracula X so what I have done is changed the BIOS directory for PCE Fast to use System card 1.0 in a different folder (still needs to be named syscard3.pce) and it works very well.

joolswills, 2016-01-31:
No intention of removing the current core.

closed 2016-02-04 by davidj1987