RetroPie/RetroPie-Setup #1196: Retropie PI and CLRMAMEPRO

egnolegno, 2016-01-20:
Hi, I need help to understand why some ROMSET (AMIDAR, ASTEROIDS as examples) working on Windows MAME emulator do not work on Retropie PI, can you help on this?
Can CLRMAMEPRO help on this?
Thanks and regards

objectchris, 2016-01-20:
it's all in the wiki https://github.com/RetroPie/RetroPie-Setup/wiki/Managing-ROMs

HerbFargus, 2016-01-20:
Github is for bug reprts, use the forum for support. 

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2016-01-20 by joolswills

egnolegno, 2016-01-20:
Why did you close this issue? On the answers there' s no solution for my issue.
Thanks

Ennio Dolfato

egnolegno, 2016-01-20:
No, not this issue

Ennio Dolfato

joolswills, 2016-01-21:
Apologies, but this issue tracker is used for retropie-setup bugs. your issue is related to romsets and the forum is a better place. cheers