RetroPie/RetroPie-Setup #1198: Add Colecovision

HerbFargus, 2016-01-21:
As there is no official ticket for this I thought I would summarise our options:

#### CoolCV

http://atariage.com/forums/topic/240800-coolcv-emulator-for-mac-os-x-linux-windows-and-raspberry/

Brilliant Mexican developer that goes by nanochess has ported his colecovision emulator to the pi.  A few users have installed it manually and have it working. He hasn't opensourced it yet (though in personal emails with him he seemed amicable to the idea at some point in the future)

#### Libretro-BlueMSX

https://github.com/libretro/blueMSX-libretro

Runs colecovision games great (just needs to be added as an option In the core options) but as far as I understand it, it lacks controller support currently.

#### ColEm

http://fms.komkon.org/ColEm/

Seems decent enough but not GPL so likely incompatible with retropie licenses.

#### Libretro-fba

Supposedly has support for colecovision but nobody running retropie has gotten it to work successfully.

http://blog.petrockblock.com/forums/topic/colecovision-5/

#### MESS

Probably not worth the effort for the amount of configuration needed.

#### AdamEm

Likely closed source so probably incompatible with retropie licenses

http://blog.petrockblock.com/forums/topic/colecovision-4/#post-111515

joolswills, 2016-01-21:
We can ship with non GPL code, and we already ship with one closed source emulator - so we could possibly ship with coolcv, but doesn't make it flexible with regards to our various platform support. The licence of the retropie-setup script is separate from the software it installs.

Would prefer a free software solution though. 

Colem requires X server I think.

HerbFargus, 2016-01-21:
My preference would be for upstream development on libretro-BlueMSX as it is already part of RetroPie and would have integrated controls. I don't know if there is any good way to map colecovision controls due to the complexity of the original controller. But until the controller hooks are coded into bluemsx, CoolCV is probably our best bet in the meantime.

HerbFargus, 2016-02-04:
Seeing as CoolCV has been added I can consider this closed. I may revisit advmess and lr-bluemsx at some future point when they become more pertinent.

closed 2016-02-04 by HerbFargus