RetroPie/RetroPie-Setup #1199: change dialog with xdialog

b4zz4, 2016-01-21:
More beautiful dialog of configuration in rasp-config and others

joolswills, 2016-01-21:
This would depend on an xserver though. It could be implemented perhaps for when running on x86 under X, but no use to the rpi etc.

b4zz4, 2016-01-21:
emulationestation dont work over Xserver?!

Jools Wills:

> This would depend on an xserver though. It could be implemented perhaps for when running on x86 under X, but no use to the rpi etc.
> 
> ---
> https://github.com/RetroPie/RetroPie-Setup/issues/1199#issuecomment-173592667

joolswills, 2016-01-21:
It does, but it is only built like that for X11 target (eg retropie-setup on a PC). On the RPI we use the framebuffer directly, which is why retropie-setup is designed for the terminal.

b4zz4, 2016-01-21:
yes. I thought it how an alternative version when it uses  a monitor and
no a terminal.

joolswills, 2016-01-21:
The other issue is that xdialog is no longer maintained and not even shipped with Debian, so it's not worth the effort unless there is an alternative that is maintained and includes all the dialog functionality.

joolswills, 2016-01-21:
There is "zenity" - not sure if it is a drop in replacement though.

joolswills, 2016-01-21:
Yeh zenity is not compatible either.. Sorry but this is a wontfix - you will have to make due with opening a terminal :)

closed 2016-01-21 by joolswills