RetroPie/RetroPie-Setup #1200: System Shutdown

partyguy2016, 2016-01-22:
The 3.4 system shutdown time is much longer than 3.3.  Is this an error?  It takes about 3 min. It was pretty much instant on 3.3.

bengus64, 2016-01-23:
I think this is because of the changes in the way the metadata is written each time you shutdown (being written to emulationstation), The more roms you have, the longer it will take. I think I have close to 12,000 total roms on my card, and a shutdown takes about 8 minutes. I could be wrong though, either way its a small price to pay for the new feature imo.

joolswills, 2016-01-23:
Yeah, it's not a bug. Before when restarting/shutting down ES got killed without shutting down correctly as it would when you "quit" meaning people lost their meta data edits etc. This is now corrected but if you have a lot of games it can take a little while.

closed 2016-01-23 by joolswills

robertybob, 2016-01-23:
@joolswills Quick question, is there a simple way for me stop any kind of saving on shutdown? All my gamelists are complete and I have no interest in number of times played or last played  data.

joolswills, 2016-01-23:
currently - press ctrl+c

joolswills, 2016-01-23:
(you might kill the script too - or you can create your own reboot script or something)