RetroPie/RetroPie-Setup #1201: Unable to shutdown Raspberry Pi 2

JerrytheB, 2016-01-22:
Hey,
when I try to shutdown from the EmulationStation menu it just restarts Emulationstation... I tried to choose the Reboot and Shutdown options, but all of them just restart ES. In 3.3 everything worked fine. 
Now I need to plug in my keyboard and shutdown from the terminal every time I need to power off my Pi. 
Is there a way to fix this, or is there another way to shutdown the Pi without plugging in a keyboard?

Thanks for the support

Jerry

joolswills, 2016-01-22:
Are you using the image or did you upgrade from an earlier version ? How did you upgrade ? Sounds like you haven't upgraded correctly.

JerrytheB, 2016-01-22:
I upgraded via Binary, from the Retropie-Setup menu.

joolswills, 2016-01-22:
Sounds like you didn't update retropie-setup first - so you have the new ES binary, without the new launch script which handles the shutdown etc.

https://github.com/RetroPie/RetroPie-setup/wiki/Updating-RetroPie

Update retropie-setup and then do the binary install.

joolswills, 2016-01-22:
Please use our forum first - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md as we use this issue tracker for bug reports (and in many cases problems are down to configuration mistakes etc - so always worth getting help first from the forum and coming here if others can confirm it as a problem)

JerrytheB, 2016-01-22:
Wow, didnt update the setup- script... Thank you! Will try it out later, this should solve my problem! 
Many thanks for the great support

joolswills, 2016-01-23:
np. cheers.

closed 2016-01-23 by joolswills